UtilitechAS/amsreader-firmware #863: Data is no longer sent to home assistant over MQTT

shadowano, 2024-11-04:
**Describe your problem**
It's been a couple of days no with no data showing up in Home Assistant. I've had this problem before, but normally this was solved by various reboots of internet, local network and AMS reader. This no longer works. Also reboot of HA server (raspberry pi) doesn't change anything

I have set up a MQTT Explorer in HA, and this connects just fine using same config as in amsreader (local IP address). I'm using a HA user for credentials. 

Trying to connect to amsreader interface doesn't work. It is unresponsive, and page doesn't load. I can only get the interface up in the browser if I run the amsreader hardware in AP mode. 

This is what I get trying to reach the amsreader when it's not in AP mode. 
[image]

MQTT has a red color, and it says MQTT: -9 in the user interface. What does this error specifically mean? I couldn't find it on the Wiki. 

**Hardware information:**
 - Country: Norway
 - Meter: Aidon
 - Encryption enabled: No
 - AMS reader:  Pow-U
 - M-bus adapter (if applicable): N/A

**Relevant firmware information:**
 - Version: 2.3.8
 - MQTT: yes
 - MQTT payload type: Home assistant
 - HAN GPIO: GPIO16
 - HAN baud and parity: 2400 8E1 (Serial conf)
 - Temperature sensors: N/A
 - ENTSO-E API enabled: no

shadowano, 2024-11-04:
After fiddling with credentials to see if amsreader would produce different errors in dashboard or mqtt broker logs, I rebooted amsreader by using the hardware button. After this, and putting it back into AP mode I can see MQTT goes from red to yellow. 

In the Mosquitto broker log file I can see that amsreader authenticate and sends data, but I can't see what is failing. 
Here is an extract from the log. 
[mqtt log.txt](https://github.com/user-attachments/files/17616745/mqtt.log.txt)

Still, no data is seen in HA, and amsleser user interface (dashboard) doesn't open.

ArnieO, 2024-11-04:
This sounds like some kind of network issue you're having. 
Can you post your config page?

Do you see any red error blinks on the device LED? https://wiki.amsleser.no/en/firmware#error-codes-on-the-led

shadowano, 2024-11-04:
Two red blinks on the device.
What specifically on the config page do you want screenshots of?

ArnieO, 2024-11-04:
> Two red blinks on the device.

OK - so the issue seems to be connection to the MQTT broker

> What specifically on the config page do you want screenshots of?

General, Network and MQTT sections are the most relevant. 

I recently saw a case that caused issues; you could check whether it could be relevant for you: 
By default, MQTT client ID is inherited from General/hostname. But the MQTT client ID seems to be more sensitive to "special characters" - which caused the issue. 

Due to this, we will from next firmware revision do more stringent filtering on the General/Hostname field.

shadowano, 2024-11-04:
[image]
[image]
[image]

ArnieO, 2024-11-04:
Debugging suggestions:

1. You're using static IP. If the IP address you use is within the range of the DHCP service in your router, the router might have allocated that IP to some other device on your network (what is referred to as an IP conflict). 
2. Could there be any other MQTT client in your network that presents itself to the MQTT broker as "mqtt_user"? It has to be a unique MQTT user name.

shadowano, 2024-11-06:
I only have 1 unit called amsleser2 in my network. On my router I have reserved the ip to the unit amsleser2. I guess I can set the amsreader device to dynamic ip instead of static. Could this setup cause any problem? I have had it for a long time.

ArnieO, 2024-11-07:
You seem to have two issues here:

**1. Difficulties reaching  to reach the amsreader when it's not in AP mode.**
You have configured the amsreader with static IP address on 192.168.68.101

**2. Difficulties connecting the amsreader to your MQTT broker**
You have configured the amsreader to find the MQTT broker on 192.168.68.121

To assess and assist I need some more information:

- Do you have other devices on your network that are connecting to the MQTT broker, working correctly?
- Is the MQTT broker also configured with static IP?

I might be wrong - but have a suspicion that your issue (in both cases) could be IP conflicts within your Wi-Fi network.

Let us first try to resolve the amsreader issue. 
Try this: 
While in AP mode: Config screen > Change from Static IP to dynamic (DHCP):
[image]
(Let the mDNS service be activated)
Click the "Save" button (bottom right) and wait for the device to reboot. You should now be able to access it by writing `amsleser2.local` in the URL field of your browser. 

If you now have contact with it, that will be an indication that the static IP was the issue. 
Go to the status/info page and check which IP address the DHCP service has allocated. 
If it is NOT 192.168.68.101: Open a command window and run the command `ping 192.168.68.101`. If you get a response, you know there is an other device on your network that has that IP address. You should be able to find it in the router control panel - and maybe identify it. 

It is good practice when using static IP to ensure that IP addresses are only allocated outside the IP range used by the DHCP service in the router. On most routers, the DHCP range can be set in the router control panel - somehow. It will be a bit different on each router.

This is the control panel of my tp-Link mesh network. The DHCP server allocates IP addresses dynamically in the range 10.0.0.100 - 10.0.0.199
[image]

This range should NOT be used for static IP addresses, as you risk that DHCP has allocated a certain address when a device connects and requests the same address. This would cause issues. 
On my router, it is safe to allocate static IP addresses in the range up to 10.0.0.99 and in the range 10.0.0.200 - 10.0.0.255.

An other issue you might have is reuse of the MQTT username. 

IF you have other MQTT clients connecting to the MQTT broker, they MUST present themselves to the broker with a unique name. In your case, your amsreader presents itself to the broker as `mqtt_user` :
[image]

If you have other MQTT clients using the same Username, it will not work. I recommend to use for instance `amsleser `here - or something else that ensures there are no duplicate names in your network.

Let me hear how it goes.

ArnieO, 2024-11-07:
Strange. The indications from your Mosquitto log could be useful to understand what is going on, but I need help from my colleague @gskjold here, maybe he kan make sense of what you're seeing (I do not use HA, so cannot verify what is normal - and not in the log).

Do you experience any other "strange issues" on your network - like other devices falling out from time to time?
Which brand and model router is this?

gskjold, 2024-11-09:
From what I can read you have a RPI with HA and Mosquitto MQTT broker. Did you install all of it from scratch, or is this a available RPI image? Also, which version RPI?

If you can find it, I would also like to know some config from your mqtt if they differ from default: keepalive_interval, max_keepalive, idle_timeout, restart_timeout

shadowano, 2024-11-11:
I'm sorry for not updating this thread earlier. The AMS device successfully connected 3 days ago, without me doing anything. I found out the day after. I did have 1 or 2 network drop-outs this day, and it's likely that this for some reason fixed the issue. I looked at my TP-Link Deco 3M system log, but the earliest log line was 1h after the AMS device connected again. 

My Mosquitto settings are all using default values. Mosquitto was installed as this was preferred by HA users. I'm not sure what you mean when you say RPI. Mosquitto was installed as an Add-on to HA. 

This is all strange, as I have rebooted both Internet modem, WIFI router and AMS device several times without success.

ArnieO, 2024-11-12:
RPI = Raspberry Pi.

shadowano, 2024-11-18:
I installed from an RPI image like this https://www.home-assistant.io/installation/raspberrypi. I have an RPI 4

ArnieO, 2024-11-18:
If I understand correctly, you still have issues?

Did you consider my advise in an earlier post that the reason for your issues could be an IP conflict? Based on your symptomes, this is not an unlikely cause. 

> You're using static IP. If the IP address you use is within the range of the DHCP service in your router, the router might have allocated that IP to some other device on your network (what is referred to as an IP conflict).

shadowano, 2024-11-18:
Well, it's working as it suddenly went online without me doing anything. It can be related to network, but not as an IP conflict. I have reserved the IP address on my WIFI router. I will have to dive into the network logs next time this happens. I think we can close this issue for now. Thanks for your help guys, I really appreciate it.

closed 2024-11-18 by shadowano