UtilitechAS/amsreader-firmware #867: Unencoded control character in JSON string in 2.3.9

cmock, 2024-11-14:
**Describe the bug**
2.3.9 produces invalid JSON for the "power" MQTT topic:

```
00000000  7b 22 6c 76 22 3a 22 22  2c 22 69 64 22 3a 22 53  |{"lv":"","id":"S|
00000010  41 47 59 05 e8 94 b6 22  2c 22 74 79 70 65 22 3a  |AGY....","type":|
```

Specifically, the 0x05 at offset 0x13. In previous versions, the "id" key was empty.

**To Reproduce**
Steps to reproduce the behavior:
1. Parse the MQTT JSON
2. See parser throw an exception

**Expected behavior**
I'm not a JSON expert, but "jq" says "control characters from U+0000 through U+001F must be escaped".

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware information:**
 - Country: Austria
 - Meter: Sagemcom T210-D
 - Encryption enabled yes
 - AMS reader: ESP32
 - M-bus adapter (if applicable): resistive divider

**Relevant firmware information:**
 - Version: 2.3.9
 - MQTT: yes
 - MQTT payload type: HomeAssistant
 - HAN GPIO: [e.g. GPIO16
 - HAN baud and parity: 2400 8E1
 - Temperature sensors: n/a
 - ENTSO-E API enabled: no

**Additional context**
Here's the complete JSON string from the power topic attached so nothing gets lost in markdown:
[ams.json](https://github.com/user-attachments/files/17745453/ams.json)

cmock, 2024-11-15:
Do I need to do anything special to enable configuration of debug mode? I can't see it in my config page... (using 2.3.9)

ArnieO, 2024-11-15:
> Do I need to do anything special to enable configuration of debug mode? I can't see it in my config page... (using 2.3.9)

You do not have this at the bottom of the Config screen?
[image]

Noschvie, 2024-11-15:
Seems to be an ESP32-C3
https://github.com/UtilitechAS/amsreader-firmware/issues/783#issuecomment-2143769912

ArnieO, 2024-11-15:
> Seems to be an ESP32-C3

In that case: Try to run debug with an older firmware, I think v2.3.4 might work. 
We have seen issues with binaries on C3, and since we are not using them in our products, finding the issue does not have high priority

Noschvie, 2024-11-15:
Version 2.2.28 is working for me.
https://github.com/UtilitechAS/amsreader-firmware/discussions/835#discussioncomment-10477141

cmock, 2024-11-15:
I notice the "SAGY..." string at the start (offset 2) of the GCM frame, whatever that may be...

ArnieO, 2024-11-16:
> I notice the "SAGY..." string at the start (offset 2) of the GCM frame, whatever that may be...

My guess is that it is from a random memory location due to a bug - but I'm just guessing. 
We'll have to wait until @gskjold finds time to look into this.

gskjold, 2024-12-05:
I think you are right, I found something that may explain it. Please try this firmware:
[esp32s3.zip](https://github.com/user-attachments/files/18023814/esp32s3.zip)
[esp32solo.zip](https://github.com/user-attachments/files/18023815/esp32solo.zip)
[esp32.zip](https://github.com/user-attachments/files/18023816/esp32.zip)
[esp8266.zip](https://github.com/user-attachments/files/18023817/esp8266.zip)
[esp32c3.zip](https://github.com/user-attachments/files/18023818/esp32c3.zip)
[esp32s2.zip](https://github.com/user-attachments/files/18023819/esp32s2.zip)

cmock, 2024-12-06:
This looks good. The "power" topic now looks like this:

```
{"lv":"","id":"","type":"","P":0,"P1":0,"P2":0,"P3":0,"Q":0,"PO":2,"PO1":0,"PO2":0,"PO3":0,"QO":0,"I1":0.17,"I2":0.22,"I3":0.11,"U1":233.40,"U2":233.30,"U3":233.50,"PF":15,"PF1":0,"PF2":0,"PF3":0,"tPI1":0.000,"tPI2":0.000,"tPI3":0.000,"tPO1":0.000,"tPO2":0.000,"tPO3":0.000}
```

No more funny characters!

closed 2025-01-12 by gskjold