UtilitechAS/amsreader-firmware #868: AMS reader v2.3.9 - Keeps Restarting - 1 to 3 times a hour

A-Wangen, 2024-11-14:
**Describe the bug**
After updating from v2.3.6 to v2.3.9 the AMS reader peeps rebooting 1-3 times a hour. 

**To Reproduce**
Nothing special needs to be done; it happens when I look at it in the web browser and other random times. 

**Expected behavior**

**Screenshots**
NA

**Hardware information:**
 - Country: Norway
 - Meter: Aidon
 - Encryption enabled [yes/no]
 - AMS reader: Pow-U+, esp32s2
 - M-bus adapter (if applicable): NA

**Relevant firmware information:**
 - Version: v2.3.9
 - MQTT: yes
 - MQTT payload type: RAW
 - HAN GPIO: N/A
 - HAN baud and parity: 2400 8E1
 - Temperature sensors: No
 - ENTSO-E API enabled: no

**Additional context**
NA

catohagen, 2024-11-14:
I had similar issues a few months ago, turned out to be the wifi power saving, if its on, try to set it to off or minimum

A-Wangen, 2024-11-15:
> Can you show me your config screen?

[image]

> I had similar issues a few months ago, turned out to be the wifi power saving, if its on, try to set it to off or minimum

Will try this

ArnieO, 2024-11-15:
Unless you have a really old router, you should deactivate 802.11.b compatibility. It might contribute to causing issues.

A-Wangen, 2024-11-18:
I have updated this now. removing the check on 802.11.b
The enable debugging, I did turn on after issues started. but turned of now. 
 
Per now this is the last time of reboot: 

  **Last boot: 18.11.2024 08:54
  Reason: 3 (3/0)**

Where can I see description for the reason?

ArnieO, 2024-11-18:
> Where can I see description for the reason?

There should be a description next to the code, like this: 
[image]

@gskjold: Can you help identify the reboot reason here?

A-Wangen, 2024-11-18:
> There should be a description next to the code, like this:

This is what I see: only a number 

[image]

ArnieO, 2024-11-18:
> This is what I see: only a number

I can see that - and you have an other code now: "1 (1/0)"
We need help from @gskjold to understand this.

A-Wangen, 2024-11-25:
@gskjold are you able to have a look at this? Do to down time of the ams in some critical times of the hour I now have the issue that I have gone from  10 KW to just over 15kw in "effektledd"

ArnieO, 2024-11-26:
OK - thank you. I'm puzzled that the clock goes red in periods - indicating an internet issue. 
I do not have any experience with this DNS server you're using:
[image]
Can you change it to **1.1.1.1** - which is one that we use as default, to see if that fixes the internet issue?

But let us try first to address the language issue. 
This should not show "Unknown", so there is something in your configuration that has been messed up:
[image]
Do you see "Load from server" in the pulldown?
[image]
If so: Click it. 
Then click it again and select the language you prefer:
[image]

You should then see some more information in the MQTT error messages.

A-Wangen, 2024-11-27:
Hi, I am not sure why I have set static IP in the AMS hence I have also put this in the rooter.  So, after updating this from static to DHCP. Everything looks to be working as it should, even the language came instant. 

Now I also see the message for a reboot, are there any wiki links for background for the different reasons? 
last was:  vbat power on reset (1/0) 

[image]

ArnieO, 2024-11-27:
> Hi, I am not sure why I have set static IP in the AMS hence I have also put this in the rooter. So, after updating this from static to DHCP. Everything looks to be working as it should, even the [...]

Wi-Fi connection is faster if you use Static IP. That will work great if you use:
- A local IP address that is not managed by router DHCP
- A valid DNS (for instance 1.1.1.1)

But DHCP works fine too - so you can just stay with that. 

No wiki on "Årsak", but "Vbat power on reset (1/0)" is normal for a startup "from scratch", or if the device has rebooted due to low voltage (can happen from time to time on Pow-U, but should be very infrequent). The indication is mainly used for debugging - if specific issues arise.

A-Wangen, 2024-11-27:
closed resolved with fix to WIFI settings.

closed 2024-11-27 by A-Wangen