UtilitechAS/amsreader-firmware #869: Help with Iskraemeco AM550

nori-modelar, 2024-11-15:
Hello team,

first of all thanks for the hard work you've put in this, i greatly appreciate it even though i can not get this to work. The irony here is that Iskraemeco is from my country (Slovenia) and it works flawlessly for people living in other EU countries, but not for us here. 

So the problem story:
-first i bought Slimmelezer+, did not work, no data
-bought Homewizard P1 meter, did not work, no data or extremely weird
-flashed Slimmelezer+ with AMSreader, does not work

Specs:

 - Country: Slovenija
 - Meter: Iskraemeco AM550 ED1.21 (picture attached)
 - Encryption enabled: provider( El. Ljubljana) said the port is open, no encryption
 - AMS reader: slimmelezer+

Firmware:
 - Version: 2.3.9
 - Played around with different settings, baud etc. nothing helped.
 - Also played around with hardware section UART0&2 and also invert etc..nothing changes. 
 
Now i did found on the web that im supposed to set the baud to 38400 for this Slovenian meter, if this is of any help
Telnet only tells: No HAN data received last 30s, single blink

Strangely I also found out on the web that some Slovenian users managed to get it working on the first try with homewizard reader, with the same meter and same electricity company here. 

Any help would be very appreciated! 

[image]

[image]

ArnieO, 2024-11-16:
There seems to be some particular issues with the payload formats used by Slovenian grid companies. 
Did you read this thread? https://github.com/UtilitechAS/amsreader-firmware/issues/615
_**"Each of the grid disti's has their own variant. So practically approx. 9 different versions in Slovenia."**_

- Do you have any documentation from your electricity company of your meter, especially the format of the data coming from it? 
- Run a [Telnet debug](https://www.amsleser.no/module/ets_blog/blog?id_post=24) to look at the raw data from the meter - and compare with the thread I linked.

TimotejHofbauer, 2024-11-17:
Managed to get data coming in, here is the section: 
I am running on MacOS, so i could not really find telnet function to save a file of the data coming in. 

Second question: No, currently not...will try to get it.  

Received valid DLMS at 30
(V) Using application data:
(V) 02 09 09 10  49 53 4B 31  30 32 30 37  38 34 36 30 
(V) 35 38 38 37  09 0B 49 53  4B 38 34 36  30 35 38 38 
(V) 37 06 00 00  01 85 06 00  00 00 00 06  00 00 00 00 
(V) 06 00 00 00  00 12 08 FE  06 00 00 00  00 06 00 00 
(V) 00 00 
(V) DLMS
(V) HDLC frame:
(V) 7E A0 61 CF  02 23 13 63  CA E6 E7 00  0F 00 04 56 
(V) 10 0C 07 E8  0B 11 07 15  1A 2F 00 FF  C4 00 02 09 
(V) 09 10 49 53  4B 31 30 32  30 37 38 34  36 30 35 38 
(V) 38 37 09 0B  49 53 4B 38  34 36 30 35  38 38 37 06 
(V) 00 00 01 8A  06 00 00 00  00 06 00 00  00 02 06 00 
(V) 00 00 00 12  08 FC 06 00  00 00 00 06  00 00 00 00 
(V) 2B 21 7E 
(V) LLC frame:
(V) E6 E7 00 0F  00 04 56 10  0C 07 E8 0B  11 07 15 1A 
(V) 2F 00 FF C4  00 02 09 09  10 49 53 4B  31 30 32 30 
(V) 37 38 34 36  30 35 38 38  37 09 0B 49  53 4B 38 34 
(V) 36 30 35 38  38 37 06 00  00 01 8A 06  00 00 00 00 
(V) 06 00 00 00  02 06 00 00  00 00 12 08  FC 06 00 00 
(V) 00 00 06 00  00 00 00 
(V) DLMS frame:
(V) 0F 00 04 56  10 0C 07 E8  0B 11 07 15  1A 2F 00 FF 
(V) C4 00 02 09  09 10 49 53  4B 31 30 32  30 37 38 34 
(V) 36 30 35 38  38 37 09 0B  49 53 4B 38  34 36 30 35 
(V) 38 38 37 06  00 00 01 8A  06 00 00 00  00 06 00 00 
(V) 00 02 06 00  00 00 00 12  08 FC 06 00  00 00 00 06 
(V) 00 00 00 00

TimotejHofbauer, 2024-11-22:
Managed to obtain some info from an employee at Elektro Ljubljana (distributer), sorry for the delay.

Communication:
[image]

Telegram per second:
[image]

Telegram 15min:

[image]

[image]

Is this enough? I am having hard times explaining to him what exactly to send over so if anything more is needed just let me know and make it specific.

ArnieO, 2024-11-22:
> Is this enough?

Yes, this seems to be it - and the "per second telegram" (it is maybe 5 seconds interval?) seems to correspond with what you sent earlier.
 
In order to proceed, we also need you to capture a "15 minute telegram". Can you do that?

nori-modelar, 2024-11-22:
The person from El.Ljubljana said it is 1s telegram and attached the picture above. And when im running telnet I believe im seeing data refreshing every second. 

Here you can see (when i run DLS translator) that we have data for the 59th second and then the 15min data, one after another...meaning 1 second refresh if i get it right. 
But in the last line you can see something i was not able to find anywhere on Github-therefore bellow i have attached another run of the code for the next quarter 15min reading. 

Data for UTC 19:45 and one second before here:
```
(V) HDLC frame:
(V) 7E A0 61 CF  02 23 13 63  CA E6 E7 00  0F 00 0A E7 
(V) 82 0C 07 E8  0B 16 05 14  2C 3B 00 FF  C4 00 02 09 
(V) 09 10 49 53  4B 31 30 32  30 37 38 34  36 30 35 38 
(V) 38 37 09 0B  49 53 4B 38  34 36 30 35  38 38 37 06 
(V) 00 00 02 50  06 00 00 00  00 06 00 00  00 00 06 00 
(V) 00 00 70 12  09 09 06 00  00 00 00 06  00 00 00 00 
(V) 13 66 7E 
(V) LLC frame:
(V) E6 E7 00 0F  00 0A E7 82  0C 07 E8 0B  16 05 14 2C 
(V) 3B 00 FF C4  00 02 09 09  10 49 53 4B  31 30 32 30 
(V) 37 38 34 36  30 35 38 38  37 09 0B 49  53 4B 38 34 
(V) 36 30 35 38  38 37 06 00  00 02 50 06  00 00 00 00 
(V) 06 00 00 00  00 06 00 00  00 70 12 09  09 06 00 00 
(V) 00 00 06 00  00 00 00 
(V) DLMS frame:
(V) 0F 00 0A E7  82 0C 07 E8  0B 16 05 14  2C 3B 00 FF 
(V) C4 00 02 09  09 10 49 53  4B 31 30 32  30 37 38 34 
(V) 36 30 35 38  38 37 09 0B  49 53 4B 38  34 36 30 35 
(V) 38 38 37 06  00 00 02 50  06 00 00 00  00 06 00 00 
(V) 00 00 06 00  00 00 70 12  09 09 06 00  00 00 00 06 
(V) 00 00 00 00  
(D) Received valid DLMS at 30
(V) Using application data:
(V) 02 09 09 10  49 53 4B 31  30 32 30 37  38 34 36 30 
(V) 35 38 38 37  09 0B 49 53  4B 38 34 36  30 35 38 38 
(V) 37 06 00 00  02 50 06 00  00 00 00 06  00 00 00 00 
(V) 06 00 00 00  70 12 09 09  06 00 00 00  00 06 00 00 
(V) 00 00 
(V) DLMS
(V) HDLC frame:
(V) 7E A0 37 CF  02 23 13 99  84 E6 E7 00  0F 00 0A E7 
(V) 83 0C 07 E8  0B 16 05 14  2D 00 00 FF  C4 00 02 04 
(V) 09 08 38 34  36 30 35 38  38 37 11 00  06 00 78 EB 
(V) A1 06 00 00  00 00 85 7B  7E 
(V) LLC frame:
(V) E6 E7 00 0F  00 0A E7 83  0C 07 E8 0B  16 05 14 2D 
(V) 00 00 FF C4  00 02 04 09  08 38 34 36  30 35 38 38 
(V) 37 11 00 06  00 78 EB A1  06 00 00 00  00 
(V) DLMS frame:
(V) 0F 00 0A E7  83 0C 07 E8  0B 16 05 14  2D 00 00 FF 
(V) C4 00 02 04  09 08 38 34  36 30 35 38  38 37 11 00 
(V) 06 00 78 EB  A1 06 00 00  00 00 
(D) Received valid DLMS at 30
(V) Using application data:
(V) 02 04 09 08  38 34 36 30  35 38 38 37  11 00 06 00 
(V) 78 EB A1 06  00 00 00 00  
(V) DLMS
(D) NOT Ready to update (internal clock 19:45:05 UTC, meter clock: 00:00:00, list type 3, est: 0)

``` 

Data for UTC 20:00 and one second after:

```
(V) HDLC frame:
(V) 7E A0 37 CF  02 23 13 99  84 E6 E7 00  0F 00 0A EB 
(V) 09 0C 07 E8  0B 16 05 15  00 00 00 FF  C4 00 02 04 
(V) 09 08 38 34  36 30 35 38  38 37 11 00  06 00 78 EC 
(V) 38 06 00 00  00 00 12 17  7E 
(V) LLC frame:
(V) E6 E7 00 0F  00 0A EB 09  0C 07 E8 0B  16 05 15 00 
(V) 00 00 FF C4  00 02 04 09  08 38 34 36  30 35 38 38 
(V) 37 11 00 06  00 78 EC 38  06 00 00 00  00 
(V) DLMS frame:
(V) 0F 00 0A EB  09 0C 07 E8  0B 16 05 15  00 00 00 FF 
(V) C4 00 02 04  09 08 38 34  36 30 35 38  38 37 11 00 
(V) 06 00 78 EC  38 06 00 00  00 00 
(D) Received valid DLMS at 30
(V) Using application data:
(V) 02 04 09 08  38 34 36 30  35 38 38 37  11 00 06 00 
(V) 78 EC 38 06  00 00 00 00  
(V) DLMS
(D) READY to update (internal clock 20:00:05 UTC, meter clock: 00:00:00, list type 3, est: 0, using clock: 1)
(D) Updating data storage using actual data
(D) Clearing hours from 19 to 20
(D) Clearing days from 22 to 22
(D) Day is not happy
(D)  - normal
(I) Saving data
(V) HDLC frame:
(V) 7E A0 61 CF  02 23 13 63  CA E6 E7 00  0F 00 0A EB 
(V) 0A 0C 07 E8  0B 16 05 15  00 00 00 FF  C4 00 02 09 
(V) 09 10 49 53  4B 31 30 32  30 37 38 34  36 30 35 38 
(V) 38 37 09 0B  49 53 4B 38  34 36 30 35  38 38 37 06 
(V) 00 00 02 57  06 00 00 00  00 06 00 00  00 00 06 00 
(V) 00 00 5D 12  09 07 06 00  00 00 00 06  00 00 00 00 
(V) 4B E1 7E 
(V) LLC frame:
(V) E6 E7 00 0F  00 0A EB 0A  0C 07 E8 0B  16 05 15 00 
(V) 00 00 FF C4  00 02 09 09  10 49 53 4B  31 30 32 30 
(V) 37 38 34 36  30 35 38 38  37 09 0B 49  53 4B 38 34 
(V) 36 30 35 38  38 37 06 00  00 02 57 06  00 00 00 00 
(V) 06 00 00 00  00 06 00 00  00 5D 12 09  07 06 00 00 
(V) 00 00 06 00  00 00 00 
(V) DLMS frame:
(V) 0F 00 0A EB  0A 0C 07 E8  0B 16 05 15  00 00 00 FF 
(V) C4 00 02 09  09 10 49 53  4B 31 30 32  30 37 38 34 
(V) 36 30 35 38  38 37 09 0B  49 53 4B 38  34 36 30 35 
(V) 38 38 37 06  00 00 02 57  06 00 00 00  00 06 00 00 
(V) 00 00 06 00  00 00 5D 12  09 07 06 00  00 00 00 06 
(V) 00 00 00 00  
(D) Received valid DLMS at 30
(V) Using application data:
(V) 02 09 09 10  49 53 4B 31  30 32 30 37  38 34 36 30 
(V) 35 38 38 37  09 0B 49 53  4B 38 34 36  30 35 38 38 
(V) 37 06 00 00  02 57 06 00  00 00 00 06  00 00 00 00 
(V) 06 00 00 00  5D 12 09 07  06 00 00 00  00 06 00 00 
(V) 00 00 
(V) DLMS

```

nori-modelar, 2024-11-22:
Totally missed that, but at least something is working. When i checked status on the physical meter and then AMS reader, this number is correct. 7925kWh

This is the comulative sum of high&low energy use (kWh) from the date when this meter was set up. 
This number updates every 15min, checked.

[image]

ArnieO, 2024-11-23:
Thank you - I'll check that 15 minute telegram later. 
Are you two @nori-modelar and @TimotejHofbauer with the same distributor (El.Ljubljana)?

ArnieO, 2024-11-23:
> This is the comulative sum of high&low energy use (kWh) from the date when this meter was set up.

That would be the sum of these two:
[image]

Which is correct, as we do not handle day/night tariffs in this firmware. 
So on meters that have separate counters for day/night tariff we add the two. 

It seems like we then just need a tweek to recognize correctly the other data points in the telegram. I know @gskjold has a busy period right now - so please be patient until he gets a chance to look into this.

nori-modelar, 2024-11-23:
@ArnieO It is just me, logged in from another account by mistake.

Seems like that yes, thanks for the help so far!

ArnieO, 2024-11-23:
> @ArnieO It is just me, logged in from another account by mistake.

No problem, I just wanted to be sure the logs and the info from the grid company was for the same meter.

mutymaster, 2024-11-28:
[image]

Hi, I'm also from Slovenia and this are my settings for AMS reader using Slimmelezer+ together with HA via MQTT.
Try to set HAN RX to UART2 :)

nori-modelar, 2024-11-28:
@mutymaster You did not clarify if this solves anything? Will try when i get back home but i believe you got green HAN only because you are at 20:30 therefore the 15min interval is received and the HAN goes green for a few seconds-otherwise nothing else is received like it should be.

ArnieO, 2024-11-28:
> HAN is GREEN all the time..

Green all the time?
[image]

mutymaster, 2024-11-28:
I am not aware of OBIS codes, I just wanted to say that it works for me and my needs of live data sampling frome the meter. Sorry for old photo - the point were the settings, not sure why HAN was offline at that moment. It sure is onilne now all the time. I have restarted it and here you go: 
[image]

ArnieO, 2024-11-28:
> I am not aware of OBIS codes, I just wanted to say that it works for me and my needs of live data sampling frome the meter.

Which is great! 👍 
Which grid company / distributor are you with?

mutymaster, 2024-11-28:
> Which grid company / distributor are you with?

Elektro Gorenjska

mutymaster, 2024-11-28:
> Green all the time? [...]

Ok, I have checked and tested it again, both HAN RX and TX should be set to UART2 so HAN is green and working all the time. I am apologizing for spamming wrong informations before.

ArnieO, 2024-11-28:
> I am apologizing for spamming wrong informations before.

No problem at all, confirmation on working setups is always interesting. And thank you for confirming your distributor is **Elektro Gorenjska**.

nori-modelar, 2024-11-28:
Yeah, i am facing same issues with the slimmelezer...seems like there is not enough power for the startup. Although i did manage to power it up without the powerbank, just plug it in really fast two times so some energy is still stored in the capacitors form the first unsuccessful power up. The settings in AMSreader do not really change anything (power db)

I am definitely buying AMSreader HW if this power-up thing can be solved. 

When it comes to your case, glad to see it working, but i did notice yes that Elektro Gorenjska is working for multiple user while **Elektro Ljubljana** is still a mystery :)

nori-modelar, 2024-11-28:
@ArnieO So which HW is the right for my use? It needs to be on WIFI. I believe it would be fair to contribute to you and @gskjold in that way!

ArnieO, 2024-11-28:
> @ArnieO So which HW is the right for my use? It needs to be on WIFI. I believe it would be fair to contribute to you and @gskjold in that way!

It is the **Pow-P1** - and you will see the recommendation if you use the Selection guide in our webshop: https://www.amsleser.no/index.php?fc=module&module=ams_selector&controller=display (We will add Slovenia to the list of known countries for Iskraemeco AM550 with a note regarding which distributors are currently covered.)

You can place order on Lectronz: https://lectronz.com/products/pow-p1-reader-for-power-meter-with-p1-port

svaliv49, 2024-11-28:
Just jumping in here, we had that discussion last year - hello All. Great seeing progress with the multi-standard country Slovenia 😠. I am on Elektro Celje - had to write my own script to get it working on a RPI4 and than in HA. But would definitely go with the  AMSreader once Elektro Celje (ECE) would be implemented 👍.

ArnieO, 2024-11-28:
> once Elektro Celje (ECE) would be implemented

It would be great if you could post example payloads (both the 5 second and 15 minute) and content definitions. We could then implement in parallell with Elektro Ljubljana.

ArnieO, 2024-12-05:
Those of you that are with Elektro Lubljana: Please see: https://github.com/UtilitechAS/amsreader-firmware/discussions/858#discussioncomment-11462856

We need to understand if payloads from meters with Elektro Lubljana work (or not) with correct bitrate setting. 
Are there maybe differences between 1- and 3-phase meters?

rvetrsek, 2024-12-05:
If I may, I have those settings, and it produces a seemingly correct output when debugging with telnet. However, the HAN indicator stays red, and no data is shown on the dashboard.

I'm with Elektro Ljubljana on a 3-phase connection.

gskjold, 2024-12-05:
Here is a test firmware for those who still have issues:
[esp32s3.zip](https://github.com/user-attachments/files/18023585/esp32s3.zip)
[esp32solo.zip](https://github.com/user-attachments/files/18023586/esp32solo.zip)
[esp32.zip](https://github.com/user-attachments/files/18023587/esp32.zip)
[esp8266.zip](https://github.com/user-attachments/files/18023588/esp8266.zip)
[esp32c3.zip](https://github.com/user-attachments/files/18023589/esp32c3.zip)
[esp32s2.zip](https://github.com/user-attachments/files/18023590/esp32s2.zip)

ArnieO, 2024-12-05:
> If I may, I have those settings, and it produces a seemingly correct output when debugging with telnet. However, the HAN indicator stays red, and no data is shown on the dashboard.
> 
> I'm with Elektro Ljubljana on a 3-phase connection.

Have you set baud rate to 38 400 and parity to 8N1?

rvetrsek, 2024-12-05:
Yes, these are my current settings:
[image]

They are the only way I could get the results I posted in the comment above (https://github.com/UtilitechAS/amsreader-firmware/issues/869#issuecomment-2506275302).

ArnieO, 2024-12-05:
> Yes, these are my current settings:

Interesting: One user with Elektro Ljubljana can read, an other cannot. 
Strange situation. Could one of you "locals" try to contact Elektro Ljubljana and find out whether there are different payloads being used? For instance single phase vs 3-phase meters?

zitko, 2024-12-08:
Hi all! I'm joining this thread as my grid company is also Elektro Ljubljana. I have POW-P1 device attached to IskraEmeco AM550 meter (exact model is AM550-ED1.21). The grid company configured P1 port on the meter to unencrypted mode.
My communication settings are set to 38400/8N1, unencrypted, not inverted.

@gskjold thank you very much for the updated firmware! I flashed my device but it doesn't seem to improve the situation. It correctly detects the meter as `Iskra` and shows the summary consumption (as was the case with 2.3.9). It does not show the current input and/or historical data.

[image]

Posting some frames I get through telnet interface below. Posting both LLC and DLMS frames just in case, I understood that LLC is just a wrapper around DLMS ...

**LLC frame**

Raw

```
E6 E7 00 0F  00 25 2E 9E  0C 07 E8 0C  08 07 13 05
1E 00 FF C4  00 02 0A 09  10 49 53 4B  31 30 32 30
37 38 35 37  36 36 36 34  39 09 0B 49  53 4B 38 35
37 36 36 36  34 39 06 00  00 02 1C 06  00 00 00 00
06 00 00 00  00 06 00 00  01 1A 12 09  20 12 01 16
06 00 00 02  1C 06 00 00  00 00
``` 

XML

```
<GatewayRequest>
  <NetworkId Value="231" />
  <PhysicalDeviceAddress Value="" />
  <DataNotification>
    <LongInvokeIdAndPriority Value="00252E9E" />
    <!--2024-12-08 19:05:30-->
    <DateTime Value="07E80C080713051E00FFC400" />
    <NotificationBody>
      <DataValue>
        <Structure Qty="0A" >
          <!--ISK1020785766649-->
          <OctetString Value="49534B31303230373835373636363439" />
          <!--ISK85766649-->
          <OctetString Value="49534B3835373636363439" />
          <UInt32 Value="0000021C" />
          <UInt32 Value="00000000" />
          <UInt32 Value="00000000" />
          <UInt32 Value="0000011A" />
          <UInt16 Value="0920" />
          <UInt16 Value="0116" />
          <UInt32 Value="0000021C" />
          <UInt32 Value="00000000" />
        </Structure>
      </DataValue>
    </NotificationBody>
  </DataNotification>
</GatewayRequest>
```

**DLMS frame**

Raw

```
0F 00 25 2E  9E 0C 07 E8  0C 08 07 13  05 1E 00 FF
C4 00 02 0A  09 10 49 53  4B 31 30 32  30 37 38 35
37 36 36 36  34 39 09 0B  49 53 4B 38  35 37 36 36
36 34 39 06  00 00 02 1C  06 00 00 00  00 06 00 00
00 00 06 00  00 01 1A 12  09 20 12 01  16 06 00 00
02 1C 06 00  00 00 00
```

XML

```
<DataNotification>
  <LongInvokeIdAndPriority Value="00252E9E" />
  <!--2024-12-08 19:05:30-->
  <DateTime Value="07E80C080713051E00FFC400" />
  <NotificationBody>
    <DataValue>
      <Structure Qty="0A" >
        <!--ISK1020785766649-->
        <OctetString Value="49534B31303230373835373636363439" />
        <!--ISK85766649-->
        <OctetString Value="49534B3835373636363439" />
        <UInt32 Value="0000021C" />
        <UInt32 Value="00000000" />
        <UInt32 Value="00000000" />
        <UInt32 Value="0000011A" />
        <UInt16 Value="0920" />
        <UInt16 Value="0116" />
        <UInt32 Value="0000021C" />
        <UInt32 Value="00000000" />
      </Structure>
    </DataValue>
  </NotificationBody>
</DataNotification>
```

gskjold, 2024-12-08:
Alright, lets try this again:
[esp32s3.zip](https://github.com/user-attachments/files/18053777/esp32s3.zip)
[esp32solo.zip](https://github.com/user-attachments/files/18053778/esp32solo.zip)
[esp32.zip](https://github.com/user-attachments/files/18053779/esp32.zip)
[esp8266.zip](https://github.com/user-attachments/files/18053780/esp8266.zip)
[esp32c3.zip](https://github.com/user-attachments/files/18053781/esp32c3.zip)
[esp32s2.zip](https://github.com/user-attachments/files/18053782/esp32s2.zip)

zitko, 2024-12-08:
@gskjold, great, thank you for your quick response and new firmware!

Flashed my POW-P1, seems it's working now (HAN indicator is green)!

This is how the dashboard looks like with the latest firmware:
[image]

Is there anything else I should check to confirm it's working?

gskjold, 2024-12-08:
Nice, just keep using it and see if you notice anything that does not seem right :)

nori-modelar, 2024-12-08:
@gskjold  It is indeed working for me as well! Great job, many thanks so far!

Import power seems to be accurate, i measured it with other device just to be sure. 

Funny I do not see any current going through though,  and I believe me and @zitko are on the same setup, except for the measuring HW atm (slimmelezer vs yours)

Second thing is that under "Import" gauge we see Iskra and NOT the sum of Low + High energy, but rather this number is just the Low energy import within the lifespan of our meters. Before your custom fw we saw the sum of both. Which metric  should be displayed here?

Third, Real time calculations are off, but again I don't know what should be the proper way of displaying them.

[image]
[image]

rvetrsek, 2024-12-09:
Hey guys, the latest firmware that @gskjold posted finally brought at least some numbers into my dashboard, which is fantastic after months of trying different things and failing! Thank you, @gskjold! 👏 

There is one weird thing. It seems that I'm exporting all my energy, which can't be because I don't have any alternative energy sources. So, is there a setting to let it know that that is the consumed energy?

[image]

zitko, 2024-12-09:
@gskjold, posting my observations after using firmware `0947d33` for about 24 hours:
- Import gauge seems correct
- not sure what should be shown directly under the Import gauge (- / kWh)?
- the kWh right to the Iskra label (under import gauge) seems off as it has drastically decreased from v2.3.9. It might be only the low energy (and not low + high), as noted by @nori-modelar
- Voltage & Amperage seem correct
- Real time calculations / Consumptions are not adding up. I assume the Hour part is ok but not the other three (Day, Month, Last month)
- Real time plot (W) seems correct
- Energy use last 24 hours is empy, so is Energy use last 30 days (kWh)

This is dashboard screenshot:
[image]

What I noticed in the debug logs (was not there in v2.3.9) are the following debugs:

`(D t:82718302ms) READY to update (internal clock 18:58:38 UTC, meter clock: 00:00:00, list type 2, est: 1, using clock: 1)`

When the clock hits 19:00, the following is reported:

```
(D t:82809254ms) Updating data storage using actual data
(D t:82809258ms) Clearing hours from 18 to 19
(D t:82809258ms) Clearing days from 9 to 9
(D t:82809269ms) Day is not happy
(D t:82809273ms)  - normal
(I t:82809274ms) Saving data
```

from there on, debugs like this are reported:

`(D t:82834289ms) NOT Ready to update (internal clock 19:00:34 UTC, meter clock: 00:00:00, list type 2, est: 1)`

My guess is that when clock hits `20:00` it will switch to `READY` debugs again...

My wild guess would be that the readings received from the meter are not stored for some reason, might be the clock discrepancy between the internal clock/meter? That would explain missing historical data (anything >1 hour)?

gskjold, 2024-12-12:
Thank you all for thorough testing!

@nori-modelar :

> Funny I do not see any current going through though, and I believe me and @zitko are on the same setup, except for the measuring HW atm (slimmelezer vs yours)

There seem to be two different payloads, one that contains current and one that does not. I don't know why some meters are different

> Second thing is that under "Import" gauge we see Iskra and NOT the sum of Low + High energy, but rather this number is just the Low energy import within the lifespan of our meters. Before your custom fw we saw the sum of both. Which metric should be displayed here?

Code type-o on my end, sorry, should be fixed now!

> Third, Real time calculations are off, but again I don't know what should be the proper way of displaying them.

Day and month real-time rely on the data in the day and month graph. If the graph is not populated, the real time calculation will also be off. The issue with the graph should be fixed in the attached version

@rvetrsek :

> There is one weird thing. It seems that I'm exporting all my energy, which can't be because I don't have any alternative energy sources. So, is there a setting to let it know that that is the consumed energy?

Code type-o on my end, sorry, should be fixed now!

@zitko :

I think most your questions got answered above. I must commend your post though, very tidy report of your findings, the way I like it!

> My wild guess would be that the readings received from the meter are not stored for some reason, might be the clock discrepancy between the internal clock/meter? That would explain missing historical data (anything >1 hour)?

The reason was a "list type" variable in the code, that indicated that these payloads did not have kWh meters in them. I have adjusted the setting of these variables to better reflect the content in the payloads. I am pretty sure we should see data now.

New version label should be 3bc40cb:
[esp32s3.zip](https://github.com/user-attachments/files/18108354/esp32s3.zip)
[esp32solo.zip](https://github.com/user-attachments/files/18108355/esp32solo.zip)
[esp32.zip](https://github.com/user-attachments/files/18108356/esp32.zip)
[esp8266.zip](https://github.com/user-attachments/files/18108357/esp8266.zip)
[esp32c3.zip](https://github.com/user-attachments/files/18108358/esp32c3.zip)
[esp32s2.zip](https://github.com/user-attachments/files/18108359/esp32s2.zip)

zitko, 2024-12-14:
@gskjold thank you very much for the new firmware and all the effort, much appreciated!

The following are my observations after using `3bc40cb` build for about a day:

- the kWh right to the Iskra label (under import gauge) now correctly shows low + high energy
- Export on the other hard is now reporting (duplicating) the Import value
- I assume that this (Import being duplicated as Export) also affects the Energy use last 24 hours as it shows Export (which I don't have at all). Not sure how this affects Energy use last 30 days ...
- everything else seems correct 👍 

Dashboard screenshot:
[image]

Another thing that's new from previous build are these debugs (for list type 3):

`(D) NOT Ready to update (internal clock 10:00:38 UTC, meter clock: 00:00:00, list type 3, est: 0)`

ArnieO, 2024-12-14:
> * Export on the other hard is now reporting (duplicating) the Import value

You mean Export kWh is the same as Import W?

zitko, 2024-12-14:
> > * Export on the other hard is now reporting (duplicating) the Import value
> 
> You mean Export Wh is the same as Import W?

@ArnieO, yes, the number under the Export gauge (Wh) is the same as the (correct) Import W.

nori-modelar, 2024-12-15:
@gskjold Been testing today but nothing has changed for me, it looks the same except for the sum under Import gauge (fixed now-correct value)

The dashboard screenshot:
[image]

gskjold, 2024-12-16:
If you all could, I'd love to get a debug session from f.ex. 12min past the hour and for as long as it stays alive. I need to analyze the difference between the data all of you get

nori-modelar, 2024-12-16:
Here is my log starting from approximately 9 min after the full hour:

[Terminal Saved Output.txt](https://github.com/user-attachments/files/18156698/Terminal.Saved.Output.txt)

gskjold, 2024-12-23:
Theres a bit of guesswork involved here, but lets try again:
[esp32s3.zip](https://github.com/user-attachments/files/18228639/esp32s3.zip)
[esp32solo.zip](https://github.com/user-attachments/files/18228640/esp32solo.zip)
[esp32.zip](https://github.com/user-attachments/files/18228641/esp32.zip)
[esp8266.zip](https://github.com/user-attachments/files/18228642/esp8266.zip)
[esp32c3.zip](https://github.com/user-attachments/files/18228643/esp32c3.zip)
[esp32s2.zip](https://github.com/user-attachments/files/18228644/esp32s2.zip)

nori-modelar, 2024-12-29:
Observations so far with the latest build:

1. With same settings the import data was wrong and nothing else was updating, even after 2 hours.
[image]

2. After only changing buffer size from 512 to 256 I could get correct import data in. But be aware that if I select any other number now, it still works, even at 512.

So overall for me it is the same as with the previous builds. 

[image]

zitko, 2024-12-31:
@gskjold, thank you very much for the new firmware. I've been using it for about a week now and everything seems to be working as expected with my meter. Great work! 
[image]
[image]

ArnieO, 2024-12-31:
@zitko : Good to hear! 

But it seems that @nori-modelar still has issues - and you are both with Elektro Lubljana?

zitko, 2024-12-31:
@ArnieO yes, I'm with Elektro Ljubljana, my meter is IskraEmeco AM550 (exact model is AM550-ED1.21).

ArnieO, 2024-12-31:
> @ArnieO yes, I'm with Elektro Ljubljana, my meter is IskraEmeco AM550 (exact model is AM550-ED1.21).

What about you, @nori-modelar? Which model meter do you have?
Could it for instance be that there is a difference between telegrams from 1-phase vs 3-phase meters?

nori-modelar, 2024-12-31:
You can see my specifications of the meter in the first post.

As for the difference between one and three phases- i do not know.

ArnieO, 2024-12-31:
> You can see my specifications of the meter in the first post.

Ah - yes of course. Sorry! So it seems to be the same model as @zitko : AM550-ED1.21

@zitko : Can you post a photo of your meter (like @nori-modelar did in his first post)? It could be interesting to see if they look identical.

zitko, 2024-12-31:
@ArnieO sure, photo attached.

[image]

ArnieO, 2024-12-31:
> @ArnieO sure, photo attached.

Thank you! So:

- @zitko and @nori-modelar have identical meters, and both are with Elektro Ljubljana - so we should expect that the payloads are identical. 
- @zitko has a working installation, using the test firmware posted here: https://github.com/UtilitechAS/amsreader-firmware/issues/869#issuecomment-2559456433
    [image]
- When @nori-modelar use the same firmware he does not get correct values. 

This is indeed strange!

@nori-modelar : IIRC you are using a Slimmelezer+. Could it be that you are facing the issue mentioned by @mutymaster in this post? https://github.com/UtilitechAS/amsreader-firmware/issues/869#issuecomment-2505594725

> The only issue I am facing with slimmelezer+ is that I need to start it with powering it over the usb-c connector and after it booted, I can disconnect it and it will work on its own over the LAN cable. Also, when the Wi-Fi signal is lost it powers off and I need to revive it again. Haven't yet figure that out.

nori-modelar, 2024-12-31:
As far as the hardware goes, yes I am facing multiple startup problems. But once i manage to get it online it works like it should. 
I dont see how this start up power issue could result in different data interpretation

ArnieO, 2025-01-01:
> As far as the hardware goes, yes I am facing multiple startup problems. But once i manage to get it online it works like it should. I dont see how this start up power issue could result in different data interpretation

I agree with you, it should not.

nori-modelar, 2025-01-05:
Just looked after 7 days and all of a sudden I can now see hourly data and data by days(graph). Absolutely nothing was changed in between as I was not even home :) So although the amsreader is online for 7 days, hourly and day data graphs started showing from 3th January onwards.... Very weird, but i hope it helps in some way @gskjold 

[image]
[image]

ArnieO, 2025-01-06:
> Very weird, but i hope it helps in some way @gskjold

Weird but surely helpful! 
What you experienced initially then seems to be isolated some kind of startup issue, and not directly an issue with the interpretation of the payload. 

Thank you for reporting!

ArnieO, 2025-01-06:
> The only thing I am missing, possibly because my meter does not provide the measurement, is the voltage and amperage reading in some of the other screenshots in this thread.

To see what is included in the payloads:

- Run Telnet debug: https://www.amsleser.no/module/ets_blog/blog?id_post=24
- Copy a "DLMS" section from the output
- Paste it into this translator: https://www.gurux.fi/GuruxDLMSTranslator  and convert to XML
- Study the output. HEX values must be converted to decimal, and decimal point position estimated. 
- Unfortunately, OBIS codes seems not to be used in Slovenia - so you just have to see if there is anything that could be current(s) or voltage(s).

nori-modelar, 2025-01-06:
@ArnieO He ( @rvetrsek  ) has the same missing components as I do. We both already posted this payloads. 

But then how is  @zitko able to get that data displayed?

ArnieO, 2025-01-06:
> But then how is @zitko able to get that data displayed?

A problem in Slovenia seems to be that there are several payload formats in use with different electricity companies (not standardized). Based on the received payloads here, @gskjold has been able to adjust the firmware to read those payloads - but we can only display the data that arrives from the meter. 

I assume the payloads @zitko receives from his meter has more data than the payloads you and @rvetrsek receive. 

To be sure there is nothing "hidden", you can enable all the tiles on your config page:
[image]

Tiles set to "Auto" will only be shown if there is data available for them, but make sure that no tiles are set to "Disabled".

[4 more replies]

closed 2025-02-03 by gskjold