UtilitechAS/amsreader-firmware #870: will not connect to wifi after install

nzxtdk, 2024-11-19:
**Describe your problem**
after installing the newest firmware to my ESP32, and connected to the wifi it makes to get to the config page, i but in my wifi details, and after hitting save and reboot, its stuck in a "loop" where its trying to connect to the wifi if i try with the newest firmware i can not get it to flash the firmware at all, only version there will flash is 2.1.16

LOG: 
[16:46:32](I) (setup)(C1) AMS bridge started
[16:46:32](I) (setup)(C1) Voltage: 0.00V
[16:46:32](D) (setup)(C1) ESP32 LittleFS
[16:46:32](D) (setup)(C1)  size: 1507328
[16:46:32](I) (WiFi_connect)(C1) Connecting to WiFi network: Hangar
[16:46:36]ets Jun  8 2016 00:22:57
[16:46:36]
[16:46:36]rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:46:36]configsip: 0, SPIWP:0xee
[16:46:36]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:46:36]mode:DIO, clock div:2
[16:46:36]load:0x3fff0030,len:184
[16:46:36]load:0x40078000,len:12600
[16:46:36]load:0x40080400,len:2724
[16:46:36]entry 0x40080594
[16:46:45]ets Jun  8 2016 00:22:57
[16:46:45]

**Hardware information:**
 - Country: [e.g. N
 - Meter: [e.g. Aidon]
 - Encryption enabled [yes/no]
 - AMS reader: [e.g. Pow-U, ESP32 etc]
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.1.16
 - MQTT: [yes/no]
 - MQTT payload type: [e.g. JSON]
 - HAN GPIO: [e.g. GPIO5]
 - HAN baud and parity: [e.g. 2400 8E1]
 - Temperature sensors [e.g. 3xDS18B20]
 - ENTSO-E API enabled: [yes/no]

ArnieO, 2024-11-19:
Hi, 
I will try to help you, but can you provide this information, please?

- Country:
- Meter: 
- Encryption enabled [yes/no]
- AMS reader: Pow-U, Pow-K or Pow-P1

gskjold, 2024-11-28:
It looks like you have a vanilla ESP32 dev board? When you flash it, make sure you use all four of the files, not just firmware.bin

closed 2024-12-05 by gskjold