UtilitechAS/amsreader-firmware #875: How is wattage calculated?

tomato20, 2024-11-26:
**Describe the bug**
How is the wattage of the phases calculated? Are the values read directly from the meter, or is is calculated from multiplying voltage by amps?

If the latter is the case, then some bad math seems to be used.

**To Reproduce**
Steps to reproduce the behavior:
None

**Expected behavior**
That the wattage is correct compared to voltage an amperage. This caused concerns on what values are correct.

**Screenshots**
[image]

**Hardware information:**
 - Country: Denmark
 - Meter: Kamstrup 382M
 - Encryption enabled no
 - AMS reader: Pow-K+
 - M-bus adapter (if applicable): N/A

**Relevant firmware information:**
 - Version: 2.3.9
 - MQTT: Yes
 - MQTT payload type: Home Assistant
 - HAN GPIO: Don't know
 - HAN baud and parity: 9600 8N2
 - ENTSO-E API enabled: No

**Additional context**
Add any other context about the problem here.

ArnieO, 2024-11-26:
Wattage per phase is only available in the user interface when read from the meter. 

The content of the payloads coming from the meters vary a lot. There are differences between countries, differences between meter brands - and sometimes even differences on the same meter brand and model within the same country (with Slovenia as a "good" example...)

In your case, you are on a Pow-K and Kamstrup Omnipower in Denmark. You can download the Interface Specification for your meter [here](https://www.kamstrup.com/da-dk/elloesninger/intelligente-elmaalere/modules/han-nve/documents). 
The default configuration in Denmark is "DLMS push 1", so you need to look at chapter 5.2 where you find a table showing which data is included in each payload transmitted from the meter, which includes:
[image]

> **Expected behavior**
> That the wattage is correct compared to voltage an amperage. This caused concerns on what values are correct.

The short answer is: In a 3-phase system you cannot expect to find P = U * I per phase. 
If you want to go in depth on this, you need to read up on the subject.

closed 2024-11-26 by ArnieO

tomato20, 2024-11-26:
Thank you for the feedback. 

I figured that it might have something to do with the meter not being a Kamstrup Omnipower but a Kamstrup 382M, which communicates differently than the Omnipower (382M via KMP).

I know you went through a lot of work to be able to support the 382M.

Best,
Thomas

ArnieO, 2024-11-26:
So it is a 382M - that is a completely different story: It is not sending lists of data, each data point has to be pulled from the meter (using the Kamstrup KMP protocol). 

But the principle is the same: We do not calculate parameters, we only give you what we can pull from the meter.

tomato20, 2024-11-27:
All right, thank you clarifying this.

Best,
Thomas