UtilitechAS/amsreader-firmware #876: Historical data for peak times last 2-3 months

MariusJS, 2024-11-26:
As most companies send their invoices for the last month, this would be a great feature to check that the HAN data corresponds with the invoice.
Something like this
[image]

ArnieO, 2024-11-26:
Interesting proposition!
(I taught we'd already received this request - but I'm not able to find it now.)

MariusJS, 2024-11-26:
I talked to Egil about it and he asked me to submit a ticket, so he might already have mentioned it?

ArnieO, 2024-11-26:
> I talked to Egil about it and he asked me to submit a ticket, so he might already have mentioned it?

Ah... that's it - I'm Egil. 😉 
Thank you for submitting your proposition.