UtilitechAS/amsreader-firmware #880: Future energy prices missing

nossaile, 2024-12-08:
Time is well after 1700 and future energy prices missing independently of ENTSO-E key, but with different error messages: 
if ENTSO-E key is blank then error message=”Price service: Exceeded API rate limit” and 
with correct ENTSO-E key then error message=”Price service: 503”.
I have not seen the "503"-message before!
Chip: esp32s2 (160MHz)
Device: Pow-P1
Installed version: v2.3.9

GaupeLnx, 2024-12-08:
Had the same with api key (503) and got Price service internal server error when removing the api key.
But it seems to work now again with the api key . Maybe a hick-up at entso-e

nossaile, 2024-12-08:
Still the same problem for Sweden SE4 at 1908.

gskjold, 2024-12-08:
Server is unable to get exchange rates for an upstream service. I have added a backup service for exchange rates, so most price areas should work again now. When the primary source is back on line, everything will work again. Guessing tomorrow morning.

mortenm22, 2024-12-08:
Is this related to the problem?
[image]

gskjold, 2024-12-09:
No, the API is still very much updated still

nossaile, 2024-12-09:
Prices for Sweden SE4 were back before midnight last night, but it looks like prices for all price regions were not back simultaneously. Why?

gskjold, 2024-12-12:
Actually not sure, it could be that it failed to load some of the exchange rates from the backup api

closed 2024-12-12 by gskjold