UtilitechAS/amsreader-firmware #884: AMS reader and HomeAssistant

obidobi, 2024-12-12:
I have an issue with accumulated data sent one time each hour.

Take Accumulated active import for example. It accumulates data between 13-14. In the AMS web interface it correctly shows this consumption for the hourly bar at 13.00

When sent to HA this data have a timestamp of 14. So in HA graphs it will be displayed as an hourly bar for 14. Since data collected between 14-15 show up in the 14.00 bar. But this data should actually show up in the 13.00 bar since it was collected between 13-14 in AMS.

AMS should really send hourly data with a timestamp of 13:59.59 for HA to show it correctly. Have anyone got a suggestion on how I get HA to show this data not shifted forward on hour?

gskjold, 2024-12-12:
This issue have been raised several times. I have looked at it before and have not found a way around this. If there is anyone out there who has the answer for this, please let me know and I will implement it.

gskjold, 2024-12-12:
Also mentioned here: https://github.com/UtilitechAS/amsreader-firmware/wiki/How-to-add-to-Home-Assistant#notes

ArnieO, 2024-12-12:
Short story: HA used the time of arrival of the message, and meters configured for HAN-NVE (Norway + some Swedish and Finnish meters) send accumulated energy only once per hour - shortly after whole hour. So HA registers the data in the wrong time slot.

obidobi, 2024-12-12:
Ah. Didn't realize the timestamp is not sent with the data but that HA uses the time the message arrive.
Need to figure out if this can be solved in HA.
Should be able to use the "AMS reader Current hour used" field. Going to try using a utility meter helper with hourly resets and see if that works.

ArnieO, 2024-12-12:
> Ah. Didn't realize the timestamp is not sent with the data but that HA uses the time the message arrive. Need to figure out if this can be solved in HA. Should be able to use the "AMS reader Current day used" field. Going to try using a utility meter helper with resets and see if that works.

It is only an issue in the countries I mentioned, when the meter only reports kWh once per hour (HAN-NVE configuration). In other countries, where kWh is reported in each telegram, this works as intended.

If you find a solution, please tell us. 
We have a large number of HA users in Norway and Sweden that would love to get this solved!

closed 2024-12-12 by ArnieO

obidobi, 2024-12-12:
Just wanna report on how I got around this issue in HA.
HA has a helper called [Integral](https://www.home-assistant.io/integrations/integration/)
So I compute accumulated data directly from the Active import instead of using Accumulated active import

obidobi, 2024-12-13:
Yeah planned to use those values first, but was not sure exactly how. When I found the Integral helper in HA that was so easy just plug and play the active effect and get accumulated energy usage. Since the effect is updated with high freq. the integral calculation should be precise.