UtilitechAS/amsreader-firmware #886: Getting price configuration right

t-m-cor, 2024-12-13:
### Discussed in https://github.com/UtilitechAS/amsreader-firmware/discussions/878

It seems that price configurations entrys are skipped if end-date < start-date.

This is a problem, as the configuration does not include year as an input (and should not IMHO).

As an example a price-entry starting 1. oct, and ending 31. mar will not(never?) be added.

I see two solutions (maybe there is some even better ?):
- issue a warning, if end-date < start-date, when saving (like if the to many decimal-digits warning), stating that it should be split into two (before/after 31/12)

Or, a little smoother/logical:
- add some logic, like “if end-date < start-date the end-date += 1 year”, before comparing to current date/time

gskjold, 2025-05-15:
The attached firmware should fix this issue:

[esp32.zip](https://github.com/user-attachments/files/20228529/esp32.zip)
[esp32c3.zip](https://github.com/user-attachments/files/20228532/esp32c3.zip)
[esp32s2.zip](https://github.com/user-attachments/files/20228530/esp32s2.zip)
[esp32s3.zip](https://github.com/user-attachments/files/20228531/esp32s3.zip)
[esp32solo.zip](https://github.com/user-attachments/files/20228533/esp32solo.zip)
[esp8266.zip](https://github.com/user-attachments/files/20228528/esp8266.zip)

t-m-cor, 2025-05-19:
Installed firmware from esp32s2.zip, and it now reports `AMS reader 309dd47`.

Did the following tests (as of 19. May), made an addition of a number that is unrealistic high (noticable), for the following time-ranges:
- from 1. may (would be 2025) - 31. may (would be 2025), added, working
- from 5. may (would be 2025) - 1. may (would be 2026), not added, failed
- from 31. may (would be 2025) - 25. may (would be 2026), not added, working

It seems it hasn't been fixed with the suggested firmwares :-/

ArnieO, 2025-05-21:
Thank you for feedback!

> Installed firmware from esp32s2.zip, and it now reports `AMS reader 309dd47`.

That is correct.

It looks like @gskjold still has some code adjustments to do!

t-m-cor, 2025-05-22:
Could you point me into the right place in the code ?

I'm not a skilled c-programmer though..

gskjold, 2025-05-22:
Here is another attempt at this problem (a5f95eb):

[esp32.zip](https://github.com/user-attachments/files/20391274/esp32.zip)
[esp32c3.zip](https://github.com/user-attachments/files/20391276/esp32c3.zip)
[esp32s2.zip](https://github.com/user-attachments/files/20391277/esp32s2.zip)
[esp32s3.zip](https://github.com/user-attachments/files/20391275/esp32s3.zip)
[esp32solo.zip](https://github.com/user-attachments/files/20391279/esp32solo.zip)
[esp8266.zip](https://github.com/user-attachments/files/20391278/esp8266.zip)

Code is found in #958

closed 2025-06-05 by gskjold