UtilitechAS/amsreader-firmware #888: No HAN data received

MrJLA, 2024-12-15:
**Hardware information:**
 - Country: Denmark
 - Meter: Kamstrup
 - Encryption enabled YES. Keys inserted
 - AMS reader: ESP8266
 
Board ESP8266 flashed with latest firmware.
AMSRreader installed correctly and running. I telnet into it to monitor the prompt which says it gets no data. 
I have connected the data pin to GPIO13.
[image]
HAN RX is set to GPIO13. Ive tried pull up check/unchecked. Ive tried different GPIO pins. And ive seen short signs of data but then disapears. 
What could be wrong? Im unsure what pins on the ESP8266 i should use for data connection to the meter

ArnieO, 2024-12-15:
Since you have the encryption keys, I assume you received them after contacting your grid company to request activation of the HAN port?

Are you sure you have connected it correctly to the 6-pin plug in the meter?
This is the pinout, looking into the meter:
[image]
- DATA_OUT to ESP8266 GPIO13
- GND to ESP8266 GND

gskjold, 2024-12-16:
I assume you got two encryption keys (Encryption and authentication) ?
Also, remember to set baud 2400 8N1.

ArnieO, 2024-12-16:
> I assume you got two encryption keys (Encryption and authentication) ? 

Information regarding encryption keys for Kamstrup Omnipower in Denmark: https://www.amsleser.no/module/ets_blog/blog?id_post=22

Sommerdal-ufm, 2024-12-20:
I have the same hardware as described above.
Suddenly my HAN indicator has turned red

ArnieO, 2024-12-20:
> I have the same hardware as described above. Suddenly my HAN indicator has turned red

It could be due to this Kamstrup Omnipower snag: https://www.amsleser.no/module/ets_blog/blog?id_post=33

If so, you can hope your meter is updated with latest firmware. If so, follow the guiding in the article. 
If that does not help, I hope you are in the Radius area. In that case a remote restart can be done relatively easily. You will have to contact them to request a remote restart.

Sommerdal-ufm, 2024-12-20:
I'm in the Copenhagen Radius area.
I have mailed Radius about the problem. Hopefully they will make a remote restart.

ArnieO, 2024-12-21:
> I'm in the Copenhagen Radius area. I have mailed Radius about the problem. Hopefully they will make a remote restart.

It should be trivial to get it done, as Radius have a quite close collaboration with Kamstrup - and as far as we know it is only Kamstrup that can do a remote restart, it cannot be done from the software the grid companies normally use to remote configure the meter.

Sommerdal-ufm, 2025-01-03:
Kamstrup have now done a reset, and HAN is green again.

ArnieO, 2025-01-03:
> Kamstrup have now done a reset, and HAN is green again.

Good to hear! 
This snag is unfortunate - but as you see, it is due to some kind of random issue with the Omnipower meters.

closed 2025-01-09 by gskjold