UtilitechAS/amsreader-firmware #892: EVN NetzNoe - T210-D : broken in v2.3.9

Noschvie, 2024-12-27:
**Describe the bug**
Upgrading from 2.2.28 to 2.3.9 breaks the HAN connection.

**To Reproduce**
Steps to reproduce the behavior:
.) install 2.2.28 using https://esp.huhn.me/
.) config HAN port (2400, 8N1, 512, encrypted
.) OTA update esp32c3-2.3.9-1453bbb
   -> not HAN connection
.) OTA update esp32c3-2.3.9
   -> not HAN connection

**Expected behavior**
Get the HAN connection working again.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware information:**
 - Country: Austria
 - Meter: T210
 - Encryption enabled yes
 - AMS reader: ESP32-C3 / David B.
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.3.9
 - MQTT: yes
 - MQTT payload type: JSON
 - HAN GPIO: 
 - HAN baud and parity: 2400 8N1
 - Temperature sensors 
 - ENTSO-E API enabled: no

**Additional context**
https://github.com/UtilitechAS/amsreader-firmware/discussions/480
#407, #488, #407

ArnieO, 2024-12-27:
I believe esp32c3-2.3.9-1453bbb is the test version with the new memory partitioning scheme: https://github.com/UtilitechAS/amsreader-firmware/discussions/879

So I think your message is very relevant feedback for that discussion - but not really relevant as a separate "Issue". 
You should be able to get up and running by doing a full installation (all four files) of v2.3.9 - right?

If you agree: Could you copy your message to https://github.com/UtilitechAS/amsreader-firmware/discussions/879
and close this issue?

Noschvie, 2024-12-27:
> You should be able to get up and running by doing a full installation (all four files) of v2.3.9 - right?

No, sorry, tried a full installation using https://esp.huhn.me.
The installation is successful and the MQTT connection is established correctly. Only the HAN connection cannot be established, although the same parameters as with version 2.2.28 are used.

ArnieO, 2024-12-27:
That is strange. Did you try a full (separate) Erase flash before reflashing?

Could potentially be more than one issue here - maybe isolated to C3 (as v2.3.9 is running without issues on a large number of devices; we have registered more than 1000 installations). 
@gskjold should look into it when he finds time.

dbeinder, 2024-12-27:
Please make sure you have checked 'inverted' serial:
[image]

In my initial pull request, there was code to force this setting for my board, but I believe those lines got removed at some point. Unfortunately I've been quite busy lately...

Noschvie, 2024-12-28:
Oh... I'll test it in the evening.

Noschvie, 2024-12-28:
Ok, "inverted" have to be selected, now the HAN connection is working, thanks again!

closed 2024-12-28 by Noschvie