UtilitechAS/amsreader-firmware #893: `rtc` field is zero in JSON

dbeinder, 2024-12-28:
**Describe the bug**
The `rtc` field is `0`. Works in 2.3.7, but always sends `0` in 2.3.9. I have also tested with af6ecc5 from https://github.com/UtilitechAS/amsreader-firmware/issues/867 which fixes `meterId` but still reports zero in the rtc field.

2.3.7
```
{"id":"70:04:1D:00:C1:B8","name":"ams-og","up":55,"t":1735407960,"vcc":0.000,"rssi":-81,"temp":-127.00,"data":{"lv":"","meterId":"1KFM0200169990","type":"","P":59,"Q":0,"PO":0,"QO":0,"I1":0.10,"I2":0.16,"I3":0.13,"U1":227.10,"U2":225.10,"U3":225.20,"tPI":1685.334,"tPO":0.000,"tQI":21.038,"tQO":1320.125,"rtc":1735407960},"realtime":{"h":0.00,"d":0.2,"t":5,"x":1.29,"he":0.00,"de":0.0}}
```

2.3.9
```
{"id":"70:04:1D:00:C1:B8","name":"ams-og","up":34,"t":1735407660,"vcc":0.000,"rssi":-81,"temp":-127.00,"data":{"lv":"","meterId":"KFM\x10 \x02\x98\x06","type":"","P":60,"Q":0,"PO":0,"QO":0,"I1":0.10,"I2":0.16,"I3":0.13,"U1":224.80,"U2":226.00,"U3":227.30,"tPI":1685.329,"tPO":0.000,"tQI":21.038,"tQO":1320.122,"rtc":0},"realtime":{"h":0.00,"d":0.2,"t":5,"x":1.29,"he":0.00,"de":0.0}}
```

af6ecc5 
```
{"id":"70:04:1D:00:C1:B8","name":"ams-og","up":511139,"t":1735407585,"vcc":0.000,"rssi":-82,"temp":-127.00,"data":{"lv":"","meterId":"1KFM0200169990","type":"","P":59,"Q":0,"PO":0,"QO":0,"I1":0.10,"I2":0.16,"I3":0.13,"U1":224.20,"U2":225.90,"U3":227.40,"tPI":1685.328,"tPO":0.000,"tQI":21.038,"tQO":1320.121,"rtc":0},"realtime":{"h":0.02,"d":0.3,"t":5,"x":1.29,"he":0.00,"de":0.0}}
```

**Hardware information:**
 - Country: Austria
 - Meter: Kaifa MA309M H4LAT1
 - Encryption enabled: yes
 - AMS reader: µHAN mosquito

**Relevant firmware information:**
 - Version: 2.3.9 and af6ecc5
 - MQTT: yes
 - MQTT payload type: JSON (multi-topic)
 - HAN baud and parity: 2400 8E1

ArnieO, 2024-12-30:
Thank you for noticing!
And "µHAN mosquito" means it is ESP32-C3 - right?

ArnieO, 2024-12-30:
Is your payload type "JSON (classic)" (not multi-topic)?
I did a test with v2.3.9, and get this from my Pow-K / Kamstrup Omnipower. The "rtc" field is not there:
[image]

dbeinder, 2024-12-30:
Yes, ESP32-C3, and I do have selected multi-topic.
I believe yours is a List2 frame, mine seems to be [List4](https://github.com/UtilitechAS/amsreader-firmware/wiki/Message-formats#list-4). A sample of the DLMS frame is here (https://github.com/UtilitechAS/amsreader-firmware/issues/809). My meter sends only this one type, every 15s.

ArnieO, 2024-12-30:
That's interesting. 
When "JSON (multi-topic)" is selected you should not get a nested JSON like those you show there, there should be separate JSONs for each topic. 

Let us wait until @gskjold gets the chance to look into this.

dbeinder, 2024-12-30:
I multi-topic mode the nested json is published to $DEVNAME/list3. In classic mode it is published directly to $DEVNAME

ArnieO, 2024-12-30:
Hah - thank you for that reminder!

With "JSON (multi-topic)", this is my List 2 (still no `rtc` field here):
[image]

From my meter I get List 3 only once per hour, I'll try to catch it when it arrives.

ArnieO, 2024-12-30:
Yes, I see the same here, with v2.3.9: `rtc` = 0
Have not tested with other versions. 
[image]