UtilitechAS/amsreader-firmware #894: Invalid characters in MQTT payload for Home Assistant

Rasfumi, 2024-12-29:
Meter ID field in JSON contains invalid characters for my Kaifa Meter.
After updating from version 2.3.8 to 2.3.9 the energy consumption measurements are not recognised by Home Assistant.

Payload with V2.3.9
{"lv":"","id":"KFM ","type":"","P":386,"Q":0,"PO":0,"QO":0,"I1":0.71,"I2":0.91,"I3":1.81,"U1":234.50,"U2":234.90,"U3":234.70}

Payload with V2.3.8
{"lv":"","id":"1KFM0200182517","type":"","P":282,"Q":0,"PO":0,"QO":0,"I1":0.44,"I2":0.88,"I3":1.78,"U1":234.90,"U2":235.30,"U3":234.30}

I reverting back to 2.3.8 solves the issue.

**Hardware information:**
 - Country: Austria
 - Meter: Kaifa
 - Encryption enabled: yes
 - AMS reader: esp8266

**Relevant firmware information:**
 - Version:  2.3.9
 - MQTT: yes
 - MQTT payload type:  JSON
 - HAN GPIO: GPIO5
 - HAN baud and parity:2400 8E1
 - ENTSO-E API enabled: no

dbeinder, 2024-12-29:
see here: https://github.com/UtilitechAS/amsreader-firmware/issues/867

ArnieO, 2024-12-30:
Same issue as https://github.com/UtilitechAS/amsreader-firmware/issues/867
Closing thread.

closed 2024-12-30 by ArnieO