UtilitechAS/amsreader-firmware #897: MQTT Password will not work with certain characters.

twh-tech, 2025-01-07:
When setting an MQTT password in the AMS Reader web interface, characters like ¤%" are allowed by the web UI, but MQTT access will be denied.
No warning is given, but it will simply never make a green MQTT connection.
Instead it will be red and it will state **MQTT: Connection denied**
I don't know if it is only one of the three characters I listed that is causing the problem, but at least one of them and possible all the SHIFT + numbers - i.e.: !"#¤%&/()=? are causing the problem.
I have no trouble connecting to the MQTT broker from e.g. MQTT Explorer with a password with those special characters.

I don't know what the problem is, but my guess is that the characters entered in the MQTT password box are not the ones used when trying to connect to the MQTT broker. Maybe somehow they are modified or lost when the password is stored in the AMS readers own memory or when the AMS Reader is trying to connect to the MQTT broker.

I use a Danish keyboard.

I propose either of the three options listed here:

1. The code is fixed so that those characters work with the MQTT password or
2. It is made impossible to use these characters in the MQTT password box in the web interface or
3. At least the user should be warned that those characters are not allowed

Many password prompts require the use of those characters (instead of only using a..z, A..Z, 0..9), so I expect that other users will try to use them as well.

It is not important for myself to have this fixed as I now avoid those characters in the MQTT password, but I did spend many hours trying many different things before I guessed what the problem was. I just want to help others avoid this problem.

**Hardware information:**
 - Country: Denmark
 - Meter: OmniPower
 - Encryption enabled: yes
 - AMS reader: Pow-K+
 - M-bus adapter (if applicable): ?

**Relevant firmware information:**
 - Version: 2.3.9
 - MQTT: yes
 - MQTT payload type: Home-Assistant
 - HAN GPIO: ?
 - HAN baud and parity: 2400 8E1
 - Temperature sensors ?
 - ENTSO-E API enabled: yes

ArnieO, 2025-01-08:
Thank you for your report!

Please note that you can check the stored MQTT password: Go to the info/status page and do a configuration backup with "Include secrets" checked, then study the content of the backup file. 

@gskjold will look into the password character filtering when he's back from vacation. 
I agree with you that there should be a warning if an illegal character is attempted used. 

Side note: 
I know from experience that this character: " (the double quotation mark) sometimes creates issues when copying code snippets from internet. I believe the reason is that Unicode U+0022 is almost indistinguishable from U+201C/U+201D (left and right double quotation mark) in some fonts. Hypothetically, password copy/paste could suffer from the same issue.

We have had one case where a user copy/pasted a password and by error got a space mark as the first character. That one took quite some effort to debug...

twh-tech, 2025-01-08:
Thanks @ArnieO. @gskjold 
I just tried saving the configuration with the MQTT password: 1234567890!"#¤%&/()= like you instructed.
It appears that it is the character ¤ (ASCII 0xA4) that is causing the problems as it will be converted to 0x20 0x20 in the configuration file (i.e. two blank spaces). On my Danish keyboard I get this character by typing SHIFT + 4.
If 0x20 0x20 is then used by the Pow-K+ to connect to the MQTT broker which expects the ¤ character (i.e. 0xA4), then obviously it will be denied access.

I have found out that the character is not present on all western keyboards (e.g. US keyboards), but it is present on (some) Nordic and possibly German keyboards.

Thanks for a great product by the way.
I'm currently obstructed by the OmniPower HAN port not working issue described on amsleser.no and I am following the instructions there and have contacted my provider (Radius in Denmark) and am hoping for an OmniPower firmware upgrade or at least a restart of my meter.

Kind regards,
Thomas.

ArnieO, 2025-01-08:
OK, I see - and I have the same character on my Norwegian keyboard. 
That character is most probably filtered out (@gskjold will be able to confirm). 
And I agree: The user should get a warning!

You should change your password on the MQTT broker to get around this.

That Kamstrup snag is a real nuisance! 
Fortunately, Radius has good relations with Kamstrup (that does the remote reboot). I am sure you will have your HAN port up and running again quite soon. 

And since we are still getting notifications about this happening from time to time, the firmware upgrade that should have fixed it seems to not do what was expected.

PS: Thank you for kind words!

closed 2025-01-12 by gskjold