UtilitechAS/amsreader-firmware #899: Pow-K+ stops sending payload to MQTT broker after few hours

yoabau, 2025-01-11:
**Describe the bug**
Referring to [bug ](https://github.com/UtilitechAS/amsreader-firmware/discussions/738#discussioncomment-9061913)described short after release of the firmware 2.3 last year
I made a new test this week in the hope that the bug was fixed meanwhile but the same problem occurs. The MQTT broker receives data but the device stops sending after few hours. The test started on the 7th of January and at 22: 28 and stopped on the 8th of January at 15:56. The LED blinks green and blue, but I can neither ping its static IP address nor access its web server. I need to unplug, wait 1min and the and plug it again before accessing web server again.

**Screenshots**
ref. attached data from MQTT broker for more info.

**Hardware information:**
 - Country: [Norway]
 - Meter: [Kamstrup, Model: 6841121BN243101040]
 - Encryption enabled [no]
 - AMS reader: [Chip: esp32s2 (160MHz), Pow-K+]

**Relevant firmware information:**
 - Version: [e.g. 2.3.9]
 - MQTT: [yes, SSL]
 - MQTT payload type: [JSON flat]
 - ENTSO-E API enabled: [yes]

[ams_3203_264_6_device_t_variable_ANTYEZKXAX.csv](https://github.com/user-attachments/files/18387429/ams_3203_264_6_device_t_variable_ANTYEZKXAX.csv)

ArnieO, 2025-01-12:
Sounds like it disconnects from your Wi-Fi?
Do you see any red error blinks on the device?
Any chance this could be due to an IP conflict?

yoabau, 2025-01-13:
@ArnieO :
I not sure if the ams reader is disconnecting from Wifi, but the LED flashes only blue and green, and not red.
There is unlikely an IP conflict, because I configured my router with 2 separate areas (DHCP and static), where only 3 devices use static IP for home automation.
But, when the ams reader is configured to communicate with my HomeAssistant's MQTT broker, there is no such communication failure. The ams reader sends data at regular interval without being disconnecting:

[image]

ArnieO, 2025-01-13:
OK, I see - and agree with you: This is unlikely to be an IP conflict. 
The issue occurs while sending data to an external (cloud) MQTT broker (Ubidots).

Considering your old bug report (https://github.com/UtilitechAS/amsreader-firmware/discussions/738#discussioncomment-9061913) I see that @gskjold did not follow up in that thread - so it is possible that he has not investigated this any further; sorry for that.

You proposed in that thread to borrow us access to your credentials at Ubidots. 
Could you email me (post@amsleser.no) those credentials? 
I will then ensure this is looked into and followed-up.

avandorp, 2025-01-13:
You could try using an MQTT bridge between the amsReader and Ubidots. The mosquitto broker e.g. has a bridge mode. Then you could easier analyze with mosquittos logs whether the amsReader stops sending MQTT messages or whether Ubidots stops accepting them.

gskjold, 2025-01-16:
After some testing I suspect this is triggered by a memory leak. At least in my case the available memory keeps reducing until it suddenly stops working.

gskjold, 2025-01-23:
I think I have found and corrected the issue, a test version here:

[esp32s3.zip](https://github.com/user-attachments/files/18521025/esp32s3.zip)
[esp32solo.zip](https://github.com/user-attachments/files/18521024/esp32solo.zip)
[esp32.zip](https://github.com/user-attachments/files/18521027/esp32.zip)
[esp32c3.zip](https://github.com/user-attachments/files/18521026/esp32c3.zip)
[esp32s2.zip](https://github.com/user-attachments/files/18521028/esp32s2.zip)

closed 2025-02-03 by gskjold

yoabau, 2025-02-10:
Hi,

I've been testing the new release and I can confirm that my AMS reader sends data uninterruptedly since last week. The issue seems to be resolved from my side. Thanks for your quick help 😉

ArnieO, 2025-02-10:
> Hi,
> 
> I've been testing the [Release v2.4.1](https://github.com/UtilitechAS/amsreader-firmware/releases/tag/v2.4.1) and I can confirm that my AMS reader sends data uninterruptedly since last week. The issue seems to be resolved from my side. Thanks for your quick help 😉

Good to hear - and thank you for the feedback!