UtilitechAS/amsreader-firmware #902: Upgrade to 2.4.0 fails - ESP32 DevKitC

drlauridsen, 2025-01-16:
Cannot upgrade from 2.3.9 to 2.4.0, fails with error -100
Tried both automatically OTA and manually uploading firmware in GUI.

Have not manually flashed it yet, will wait for reply here..

It is an ESP32 DevKitC, OTA has always worked.

It looks like the partitioning might have changed in 3.4.0 - is that the reason? 

[image]

ArnieO, 2025-01-17:
Already reported in https://github.com/UtilitechAS/amsreader-firmware/discussions/900, where we collect feedback on v2.4.0. 
Manual flashing will work, you should then be able to continue to the next versions with OTA.

closed 2025-01-17 by ArnieO

drlauridsen, 2025-01-17:
Thanks, I did look through all issues and did not find any relevant posts - apparently it was under discussions instead.