UtilitechAS/amsreader-firmware #904: AMS reader cloud binding not possible

mateblue, 2025-01-17:
Hi
I know: AMS reader cloud functionality is "Work in progress, functionality is being developed."
I'm not sure if it should already work?

**Describe the bug**
When I try to setup then "AMS reader cloud" and click "Connect device to my cloud account" 
the PopUp says: Not able to bind to cloud

[image]

Debug says: URI '/cloudkey.json' was not found

**To Reproduce**
Steps to reproduce the behavior:
1. Go to /configuration page
2. Cloud connections -> Enable "AMS reader cloud" (select any Protocol)
3. Save
4. Go to /configuration page again
5. Click on "Connect device to my cloud account" (under Cloud connections
6. "Not able to bind to cloud" apears

**Expected behavior**
New Tab opens and asks for login to the cloud

**Screenshots**
[image]

**Hardware information:**
 - Country: Switzerland
 - Meter: Landis & Gyr
 - Encryption enabled no
 - AMS reader: ESP32c3

**Relevant firmware information:**
 - Version: v2.4.0
 - MQTT: yes
 - MQTT payload type: json
 - HAN GPIO: GPIO7
 - HAN baud and parity:  2400 8E1
 - Temperature sensors: none
 - ENTSO-E API enabled: no

**Additional context**
Add any other context about the problem here.

ArnieO, 2025-01-17:
Thank you for your thorough and well-formulated feedback!
Amsreader cloud is working, I have it running on my Pow-K with v2.4.0. 

I note that you are running this on an ESP32-C3, and we have already seen some minor issues with the firmware running on C3. 

This is interesting:

> Debug says: URI '/cloudkey.json' was not found

It looks like a bug. Potentially isolated to C3, since we do not see the issue on S2 (that we use in our products).

gskjold, 2025-01-18:
I see the error, my mistake... I will fix this in a upcoming patch. Looks like we will need a 2.4.1 patch for some smaller snags.

ElMyggo, 2025-01-18:
I sucessfully upgrade from 2.39 to 2.40 on my ESP32 WROOM module (with flashing files and restore config) and i have to test the cloud.... and it works.
I should test it more when im not home...

gskjold, 2025-01-23:
How about this one:
[esp32c3.zip](https://github.com/user-attachments/files/18521026/esp32c3.zip)

mateblue, 2025-01-23:
Wohooooo! 
Thanks, installed it an works!
Installed version: bf0e1d1
I flashed OTA the firmware.bin from your esp32c3.zip.
Let me know if I should test any specific cases.
My amsreader is just a few minutes up (since Flashing). I can update you after a few hours if you wish.

gskjold, 2025-01-24:
Good to hear! Let me know if you see anything strange

mateblue, 2025-01-27:
Thanks for all the good work!
Everything is still fine. Works perfectly, no issues.
It is connected to amsleser.cloud and reports everything there.
amsreader uptime >4days
( I also use MQTT (Home-Assistant Payload) with HomeAssistant and no issues - havent seen any invalide characters (like d')

closed 2025-02-03 by gskjold