UtilitechAS/amsreader-firmware #910: add solar inverter

gereic, 2025-02-03:
Hi,
First of all, I want to say "what an amazing project".
Thanks to all.

I live in Austria and use this project for our energy-community. In Austria you can build a energy community where some members producing energy with solar, wind, or whatever and on the other hand, there a customers, which uses the energy, which is not consumed by the producers itself.
This way, you can save costs and it is a win-win-situation for producer and consumer in the energy-community.

I have build a small platform in HA, where every member can see, how much energy is produced and how much energy is still avaiable.

For this I use the amsreader-firmware to read the energy-meters and it works like a charm.

The idea is also to show the produced energy in the platform (actual solar-power [W]).

So the Feature-request is a setting, where I can choose the solar-inverter and the amsreader-firmware send also this data over mqtt.
There Inverters should be selectable:

- Victron
- Solar-Edge
- Huawei
- Fronius

All of this inverters can be readout via modbus-tcp.

I have already done such a readout and can also help with integration.
What I don't know, is the following:

- where can I add such a feature
- where do I have to define settings
- where do I have to integrate the settings for the webinterface
- .....

Thanks

ArnieO, 2025-02-03:
I'm a little confused by your request. 
As far as I know, all smart energy meters have a separate counter for "negative" power and energy, meaning the surplus that is delivered back to the grid. You activate the display of surplus power on your amsreader by going to config > Meter > Production, where you enter your installation peak power. Also check that User interface > Export gauge is set to Auto or Enabled.

What you will then see is not the **produced** power and energy, but the power and energy delivered back to the grid. 

To display the **produced** power/energy, you'll need to read that information from the inverter and present it to the users in Home Assistant. I do not foresee that we integrate that in the amsreader firmware.

gereic, 2025-02-03:
It's a shame, that would be exactly what I need.
Now I need to install additional hardware.
But thanks for the answer.

closed 2025-02-03 by gereic

ArnieO, 2025-02-03:
I understand you're sorry - but we just do not want to go down that road: Pulling data from other units on your network. 

Thank you for your understanding.

gereic, 2025-02-03:
If you are interested, I designed a case for din-rail which is very useful and no need of 5V power-supply is needed.
https://www.thingiverse.com/thing:6936703