UtilitechAS/amsreader-firmware #912: Show CENT instead of EUR

Belaial, 2025-02-04:
In a previous issue ( https://github.com/UtilitechAS/amsreader-firmware/issues/834#issuecomment-2309690600 ) it was briefly mentioned that the possibility to show CENT instead of EUR could be implemented.

> Also, for NOK/SEK/DKK I have added subunit (øre) as an option. I guess we could to the same for EUR, so that when the price is really low it would show cent/kWh. Any thoughts? 

 _Originally posted by @gskjold in [#834](https://github.com/UtilitechAS/amsreader-firmware/issues/834#issuecomment-2309690600)_

I'm not sure it ever was? I have read the documentation and change logs but see no mention of it so just curious if it was ever implemented or is planned to be?

**Hardware information:**
 - Country: Finland (Belonging to SE3 price region)
 - Meter: Kamstrup
 - Encryption enabled: No
 - AMS reader: esp8266
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.4.1
 - MQTT: Yes
 - MQTT payload type: Home Asistant
 - HAN GPIO: GPIO5
 - HAN baud and parity: 2400 8N1
 - Temperature sensors
 - ENTSO-E API enabled: No

ArnieO, 2025-02-04:
> I'm not sure it ever was?

Maybe just result from lack of feedback on "Any thoughts?"
Should not be difficult to implement if our EUR users prefer the price shown in Euro-cent. 
It sounds logical to me.

Belaial, 2025-02-05:
> Maybe just result from lack of feedback on "Any thoughts?"

Maybe so, not sure, both you @ArnieO and me replied in the older thread

> Please do if it's not to much work, it might be easier to read instead of `0.0255 EUR/kWh` like it is now
> 
> 
> 
> [image]
> 
> 
> 
> Also makes it more consistent I guess, if other currencies shows (øre) and so on that also Euro would behave the same.
> 
>  

 _Originally posted by @Belaial in [#834](https://github.com/UtilitechAS/amsreader-firmware/issues/834#issuecomment-2309783911)_

> Good idea! 👍  

 _Originally posted by @ArnieO in [#834](https://github.com/UtilitechAS/amsreader-firmware/issues/834#issuecomment-2309826911)_

ArnieO, 2025-02-05:
It is just forgotten, will come in next update.

Belaial, 2025-02-05:
Sounds good!

gskjold, 2025-02-13:
It is implemented in the GUI. For Home-Assistant, we are always sending the main currency. In my opinion, it is up to HA to convert this to a subunit, as it would not know what currency this is if we specified "cent" or "øre". But I am open for input on this

Belaial, 2025-02-14:
Ah ok, yes I agree that sending main currency is the correct thing to do with HA, my misunderstanding then since I though HA would automatically show "cent" somehow, case closed!

closed 2025-02-14 by Belaial