UtilitechAS/amsreader-firmware #914: Wrong graphical display of current during export

manfi001, 2025-02-07:
**Describe the bug**
Graphical display of current wrong while exporting

[image]

**Hardware information:**
 - Country: Austria
 - Meter: Sagemcom
 - Encryption enabled yes
 - AMS reader: Wemos D1 mini
 - M-bus adapter (if applicable): yes

It's only an optical issue, but shouldn't the current appear in the negative range while exporting?

ArnieO, 2025-02-07:
Good question! Ping @gskjold

gskjold, 2025-02-13:
This question pops up from time to time. Only a few meters report negative current. Also, if you dig in the closed issues, you'll find someone arguing that current can never be negative. Considering this, we should maybe remove the negative side of that graph and only show absolute value here. My main argument for this would be that there are so few meters reporting negative current (maybe only one), so the graph is more confusing than helpful.

gskjold, 2025-02-13:
Here: https://github.com/UtilitechAS/amsreader-firmware/issues/764#issuecomment-2047072106

ArnieO, 2025-02-13:
I tend to agree with what @tronde-ams writes in that referenced post - and think our approach should be to always show amperage as absolute values. This ensures consistency for all meter types, including those few that actually indicate a "direction" of the current per phase.

closed 2025-02-13 by gskjold