UtilitechAS/amsreader-firmware #918: AMS Reader Reconnection Issue After MQTT Server Restart

VesleSlotten, 2025-02-10:
**Describe the bug**
If MQTT stops and then restarts, it seems that the AMS reader does not reconnect. As far as I can see, this issue appeared after upgrading to 2.4.1 (it might have been there earlier, but I hadn't noticed it. The MQTT server is not restarted often).

**To Reproduce**
I can easily reproduce this by stopping the MQTT server, where you correctly see an error message in AMS Reader "MQTT: Connection failed".

If you then restart the MQTT server, the AMS reader does not reconnect. Other services like Home Assistant reconnect as expected (this can be seen in the Mqtt log).

**Expected behavior**
I expect the AMS Reader to reconnect to the Mqtt server when it becomes available. In the tests I have done, the MQTT server has not been unavailable for more than a couple of minutes, maximum 5.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware information:**
 - Country: Norway]
 - Meter: Aidon
 - Encryption enabled no
 - AMS reader: ESP32 
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.4.1
 - MQTT: yes
 - MQTT payload type: JSON Classic
 - MQTT: MQTT server is secured by SSL
 - HAN GPIO: GPIO16
 - HAN baud and parity: 2400 8E1
 - Temperature sensors NO
 - ENTSO-E API enabled: yes

**Additional context**
The only significant change that has been made is that after upgrading to 2.4.1, I had to create new certificates for the AMS Reader (the backup did not include these).

ArnieO, 2025-02-10:
A large portion of our users use MQTT to connect to smart home system, there are surely more than 1000.
More than 400 users have upgraded to v2.4.1, but we do not know how many of those use MQTT - but surely the majority. 

And this report is the first issue of this kind - so I think we must start the fault-seeking by looking into the details of your situation. 

Can you share your config screen (blank out any information you consider sensitive)?

VesleSlotten, 2025-02-10:
[image]

Well, it's absolutely possible that there could be something wrong with my configuration. And I can troubleshoot on my own. I just thought it was worth mentioning in case others were experiencing the same issue.

I have tried to upload a screenshot of my configuration anyway.

I can also mention that I am running Mosquitto version 2.0.20.

ArnieO, 2025-02-11:
> I have tried to upload a screenshot of my configuration anyway.

I can't see any obvious issues here - but here are some things you could look into:

- One relatively common mistake is to use the same MQTT Client ID on multiple devices, but since you're using "ams2mqtt" I assume it is unique for this device. 
- I recommend to deactivate "802.11b legacy rates", it is only needed if you have a very old router - and may cause issues.
- What hardware are you running this on? "ESP32" indicates that this is not our hardware. Inadequate power decoupling is a real issue on devices that are so heavy on Wi-Fi exploitation as this one.

jtrosby, 2025-02-12:
I have a similar problem with v2.4.1. I haven't enabled SSL for mqtt yet because I've had problems with my meter delivering enough power even without ssl enabled. After upgrading to v2.4.1 ssl for mqtt turned on by itself in the config. Turning it off just leads to it switching back on by itself every time. Downgrading to v2.4.0 removes the issue

VesleSlotten, 2025-02-13:
Thank you. I am running on an ESP32, where I have assembled the necessary components myself. I have my own power source for this (not drawing from the HAN port). It has worked excellently so far, but it could definitely be something with the WiFi. Thanks for the help, and thanks for the work put into the product.

gskjold, 2025-02-20:
I have actually experienced this myself this weekend, when I did an upgrade of the home automation in one of my locations. Will investigate

ArnieO, 2025-02-25:
Bugfix will come in v2.4.4

commit 37aa6ae8 (2025-03-24): Fix: #918 - MQTT/SSL does not reconnect after disconnect (#933)
closed 2025-03-24 by gskjold