UtilitechAS/amsreader-firmware #920: MQTT SSL turns on by itself after upgrade to v2.4.1

ArnieO, 2025-02-12:
> I have a similar problem with v2.4.1. I haven't enabled SSL for mqtt yet because I've had problems with my meter delivering enough power even without ssl enabled. After upgrading to v2.4.1 mqtt ssl turns on by itself in the config. Turning it off just leads to it switching back on by itself every time. Downgrading to v2.4.0 removes the issue 

 _Originally posted by @jtrosby in [#918](https://github.com/UtilitechAS/amsreader-firmware/issues/918#issuecomment-2653566824)_

ArnieO, 2025-02-12:
@jtrosby: Can you go to the info page and create a config backup file (without secrets) and post it here?

jtrosby, 2025-02-12:
[configfile before downgrade.cfg.txt](https://github.com/user-attachments/files/18768876/configfile.before.downgrade.cfg.txt)

I have downgraded to 2.4.0 since this backup was made. The mqtt ssl checkbox remained enabled after the downgrade, but unlike on v2.4.1 I was able to change the config to turn it off

closed 2025-02-12 by ArnieO