UtilitechAS/amsreader-firmware #924: Tariff peak improvements

mrsparx, 2025-02-17:
**Is your feature request related to a problem? Please describe.**
Knowing when my tariff peak is most common to be able to make electricity cost lower.

**Describe the solution you'd like**
A list of 10 highest tariff peaks with information about day of week and also time of day. Perhaps even what phase it was?
Or simply mouse-over information of the three peaks currently.

**Describe alternatives you've considered**
The only alternative is to login every day to amleeser to see if the peak has moved for the day.

ArnieO, 2025-02-17:
Are you looking for a visual indication on the main screen, or are you using using MQTT to read data into Home Assistant or an other Home Automation system?

mrsparx, 2025-02-17:
Hi! I do use both, but I usually end up going back to the main interface of the reader itself since it is so simple and has everything easily readable. Even though its not so pretty its very functional.

mrsparx, 2025-02-17:
I would prefer it to be available to the main page of the amsreader firstly.  MQTT is not really a priority for me. 

No I think we are talking about different things now. Im talking about the tariff peaks. Where the average of the 3 highest consumptions in kWH are used to charge the customer for high momentary kW consumption.

ArnieO, 2025-02-18:
> No I think we are talking about different things now. Im talking about the tariff peaks. Where the average of the 3 highest consumptions in kWH are used to charge the customer for high momentary kW consumption.

OK - that is something else. You're talking about the calculation of the Power tarriff (Norwegian: _**effekttariff**_). 
In Norway this is calculated from **the average of the three highest-consumption hours in 3 different days of a month**.

In the User Interface there is a separate graph for tracking that, here is my current graph:

[image]

It shows my status so far this month:
- Highest consumption hours on 11th, 25th and 17th Feb. 
- The average is 9.2 kW
- The next treshold I risk passing is 10 kW (red line)

You need to configure the tresholds to fit with how your grid company charges you: https://wiki.amsleser.no/en/firmware#tariff-thresholds

mrsparx, 2025-02-18:
Yes thats the one. I have the configuration I need. This feature add was to have more information to be able to adapt my own consumption to get the effekttariff lower. ( Same name in Sweden BTW).
In Sweden there is talk of going to 15minute intervals instead of 1hour also, but Im not sure how this affects the way the effekttariff is calculated. 
So I would basically want to have a top 10 list of the highest consumption hours with date, time of day etc. To see if there is a commonality between the top figures.. ( Ie cooking, car charging, cleaning and heating etc ) on a specific time of day to be able to offset some of the load to a different part of the day.

[image]

This is my own. Since one of the peaks was last day I can check the time and see that it was around 2000 yesterday that one peak was. But the others are too far back so I dont know the time of day.

mrsparx, 2025-02-18:
Perhaps some function to see more information about each of the 3 ( or prefferably more than 3 ) top peaks where the 24h graph is shown. Like clicking on the 7,64 in my case would show this:

[image]

mrsparx, 2025-02-18:
Yeah thats what I was after more or less. A mouse over is a pretty good start. I tried adding more than 3 peaks, but the limit seems to be 5. And at 5 the days go in to each other. 

Ideally there would be a whole row with the 5 higest tops for the month like this:

[image]

ArnieO, 2025-02-18:
> I tried adding more than 3 peaks, but the limit seems to be 5

We'll look into that!

And I notice the selector seems to accept up to 255, I think it should be limited to a lower number:

[image]

Sorry, but I don't understand your illustration. If the Tariff peak graph had shown the five highest consumptions on different days, wouldn't that be what you seek?

mrsparx, 2025-02-18:
So the whole point of this issue is to be able to find information about when my house consumes most power to be able to lower my cost of the bill. And then I need to know _when_ it consumes most. What day isnt enough. I need to be able to adjust my habits to move the consumption to another part of the day. So i need to know my highest peaks, what day they were. And what hour they were. Im not really sure how to explain better. 
In the illustration picture I have my 5 highest consumptions and when they occured, and I can also see what my house consumed the hour before and the hour after the peaks.

tronde-ams, 2025-02-18:
Another improvement I would like, is to show a bar for current hour consumption besides the tariff peak bars.

This will make it a lot easer to see if I need to reduce consumption elsewhere, especially while cooking food.
I don't think it is necessary to show the value, but some kind of a bar would be helpful.

[image]

ArnieO, 2025-02-19:
@mrsparx : OK, thank you for the additional explanation. 

@tronde-ams : I understand and agree. It would be useful to have something that gives the user an indication **_"If I do not take action now, the current hour could be one of the highest-consuming this month"_**.

mrsparx, 2025-02-20:
One more feature change on this item. Tell me if it needs to be in a different feature. 
In my area the effekttariff is only charged during workdays daytime 7-19. Effekttariff isnt used or counted at night at all. So the peaks should have a setting for when they are used. I know of other areas in Sweden where the effekttariff is halved during off hours. And on top of that there are different costs calculated depending on month. Winter and summer price.

ArnieO, 2025-02-20:
> One more feature change on this item. Tell me if it needs to be in a different feature. In my area the effekttariff is only charged during workdays daytime 7-19. Effekttariff isnt used or counted at [...]

Thank you for the input - and all grid cost issues are obviously (as shown in your message) a complex topic. In most countries there are a large number of grid companies (some quite small), and the tariff models are adapted to the cost structure of each grid company. The result is a "jungle" of tariff models, making it really challenging to parametrize in our code.

It is fine to have your input in this thread, as it is linked to "Tariff peak" issues. We just had an internal discussion on potential improvements (this thread and https://github.com/UtilitechAS/amsreader-firmware/issues/876) - and will look into it and attempt to make some useful improvements.

gskjold, 2025-02-20:
Regarding the request in the original post, we will not implement this feature on the device. The reason for this is that we do not store enough details to accomplish this. We are very careful about the real time data we store on the device to avoid prematurely exhausting the lifetime of the flash storage, and this level of detail is not worth the extra wear.

The change proposed by @tronde-ams is very much achievable since it is data we have readily available and can be added in an upcoming version. 

Very interesting details about the Swedish tariff @mrsparx, I guess we will have to adjust our code in the future to deal with this.

gskjold, 2025-02-20:
Agree, this is a nice feature with very little impact on storage

andreas1974, 2025-11-20:
There is a problem with the dates and times for tariff peaks, using a web browser in Windows 10:

[image]

Chrome and Firefox are the same. Zoom level does not matter either. (The dates and times show up fine when I use my mobile phone's Chrome web browser, only not on my computer.)

I also have a question regarding the tooltips mentioned in this thread:
> I have put the hour of the peak in the tooltip
I wonder how to show the tooltip. Nothing shows up when I hover the mouse pointer over the bars using Chrome on my computer, and nothing happens if I tap or hold the bars on my phone (also Chrome).

andreas1974, 2025-11-21:
> Strange. I have the tooltips both on Firefox and Chrome on Win 11.

Indeed very strange. I changed the default "popups" setting from Blocked to allowed, and also allowed unsafe element, and anything that was blocked by default in the Chrome settings, but there was no change (tooltips are not popup, I know, but just to eliminate as much as possible). I also deactivated an addon that was active, but no change regarding the tooltip. Reloading after every change.

If you right-click the tooltip and send me a screenshot of a piece of the surrounding HTML elements, then perhaps I could check for the same elements in my web browser, and look for a specific reason for the absent tooltips.