UtilitechAS/amsreader-firmware #925: Price data erased after firmware upgrade

Cartoon-Dude, 2025-02-17:
After upgrading the firmware, the price data is erased from the device, and must be manually entred again.
Tested when going from 2.4.1 to 2.4.2 to 2.4.3
esp8266

ArnieO, 2025-02-18:
So the loss of config was not when upgrading from v2.3.x to 2.4.x? 

It is always a good idea to backup the configuration each time you make changes to it.

Cartoon-Dude, 2025-02-18:
No, every other configuration is still there, only the electricity price is gone.
Did a backup, but the electricity price isn't included in it.

ArnieO, 2025-02-19:
Sorry, I don't understand what you mean. 

Electricity prices are not stored on the device, they are read from internet.
You activate "Enable price fetch from remote server" and select price region and currency. 

There are 3 lines in the backup file that store that configuration information. In my case they look like this:
```
priceEnabled 1
priceArea 10YNO-3--------J
priceCurrency NOK
```

martinkaltoft, 2025-02-19:
I am having the same issue with an ESP8266. After upgrade from 2.4.1 to 2.4.3 the price modifier configuration was missing. I backed up my configuration before the upgrade, which contained the price modifier information such as this:

`priceModifier 0 TariffLavVinter import add 0.1465   01-01 31-03
priceModifier 1 TariffhøjVinter import add 0.4396   01-01 31-03
`

But when restoring this configuration, the price modifiers were still not applied, and I now have to enter them manually again.

ArnieO, 2025-02-19:
Thank you for weighing in, @martinkaltoft . 
This seems to be a bug that we need to fix.

gskjold, 2025-02-27:
The fact that the price data is erased is because of #926

Currently implementing import and export of price modifiers through the config file, this was only half finished from before.

commit 8a35346f (2025-03-24): Fixed backup and restore of price modifiers (#925) (#932)

* Fix: #925 - Backup/restore of price modifiers

* Removed unused importy
closed 2025-03-24 by gskjold