UtilitechAS/amsreader-firmware #928: Support higher resolution of prices

gskjold, 2025-02-27:
Some countries are moving to 15min price resolution, a new implementation is needed to support this.

nossaile, 2025-03-13:
Starting June 11, 2025, spot price trading on the Nord Pool will move to 15-minute intervals

wknutsen, 2025-03-25:
Is this related why future energy prices is not correct in AMS reader lately for NO1 ?

[image]

Picture from my energy provider:
[image]

I expected a jump in price 18:00 in AMS reader

I checked entso prices and they match my energy provider, but not AMS

[image]

Seems like they have switched to 15min prices on website atleast, not sure this is also the case for API.

ArnieO, 2025-03-25:
Yes, Nordpool (that sets the prices in the Nordic market) are preparing to switch to pricing per quarter hour. Our Entso-E API has been modified to take this into account. On my amsreader in NO-3, the prices in the GUI are correct when I compare with Entso-E.

How have you configured your reader? This is the recommended setting, _**without**_ Entso-E API key:

[image]

wknutsen, 2025-03-25:
Thank you. Works great without API key

ArnieO, 2025-03-25:
> Thank you. Works great without API key

Great! 
And thank you for the heads-up that there is an issue for "old customers" that are still using Entso-E API key.

nossaile, 2025-03-25:
For SE4 price fetch works both with and without ENTSO-E API key

ArnieO, 2025-05-15:
> Currently at 1710 price fetch for SE4 does not work either with or without ENTSO-E API key. Without you get the following error message:
> 
> [image]

Yes, Entso-E had a problem earlier today, and were delayed with the publishing of prices for tomorrow. 
If you reboot your device now, it will fetch the prices.

nossaile, 2025-05-15:
I have just now rebooted both with and without ENTSO-E API key present and still no prices: for tomorrow for either.

ArnieO, 2025-05-15:
That is strange. I rebooted mine before I posted my comment at 17h24 - and got the prices for tomorrow. (Without Entso-E key.) 

What happens now if you:
- Delete the Entso-E key
- Press Save
- Reboot
?

nossaile, 2025-05-15:
I did as you suggested above and the error message disappeared, but still no prices for tomorrow. After a refresh tomorrow's prices finally appeared.

rogerthn2019, 2025-09-30:
I do have data in HA

[image]

rogerthn2019, 2025-09-30:
I guess that both [HACS](https://github.com/custom-components/nordpool) and the official [integration](https://www.home-assistant.io/integrations/nordpool/) does this :-(

closed 2025-12-04 by gskjold