UtilitechAS/amsreader-firmware #934: No active power per phase published

fimholz, 2025-02-28:
**Describe your problem**
The reported payload for the active power reading of individual phases is always zero.

Recently I got an update of the parameterization of our L&G E450 meter. Total active power, accumulated active energy and the individual currents and voltages were present before the update. The columns 12,13,14 of the first table below (active power per phase) are new. As far as I can see from Telnet debugging, the DLMS frame looks fine and holds these values.

I'm not sure if this is a bug or just not implemented for this type of meter and/or parameterization. I gathered information from amsleser.no Wiki [(data.json)](https://wiki.amsleser.no/en/api/http-json#latest-payload-from-meter) and used [Gurux DLMS Translator](https://www.gurux.fi/GuruxDLMSTranslator) to inspect the HEX data. You can find these snippets below the tables. I shortly tried the Hex dump payload over MQTT but was not able to receive any data (Home Assistant, Node-RED, MQTT Explorer) or find further documentation. If this is the way to go I'd be happy if you would give me an entry point.

## parameterization sets
### 5-seconds interval
This is the parameterset of interest.
| # | Logischer Name; Attributindex | Register
| -- | -- | --
| 1 | 0-8:25.9.0; 2 | Objektliste Push-Einstellungen Verbraucherinformation 1
| 2 | 0-8:25.9.0; 1 | Logischer Name Push-Einstellungen Verbraucherinformation 1
| 3 | 0-0:96.1.0; 2 | Geräteidentifikation 1 (Herstellerseriennummer)
| 4 | 1-0:1.7.0; 2 | Wirkleistung Bezug +P
| 5 | 1-1:1.8.0; 2 | Wirkenergie Bezug +A (QI+QIV)
| 6 | 1-0:31.7.0; 2 | Strom L1
| 7 | 1-0:51.7.0; 2 | Strom L2
| 8 | 1-0:71.7.0; 2 | Strom L3
| 9 | 1-0:32.7.0; 2 | Spannung L1
| 10 | 1-0:52.7.0; 2 | Spannung L2
| 11 | 1-0:72.7.0; 2 | Spannung L3
| 12 | 1-0:21.7.0; 2 | Wirkleistung Bezug +P L1
| 13 | 1-0:41.7.0; 2 | Wirkleistung Bezug +P L2
| 14 | 1-0:61.7.0; 2 | Wirkleistung Bezug +P L3
### 1-minute interval
These values are not being reported as well (MQTT payload nor data.json), but that would go beyond the limits of this issue.
| # | Logischer Name; Attributindex | Register
| -- | -- | --
| 1 | 0-9:25.9.0; 2 | Objektliste Push-Einstellungen Verbraucherinformation 2
| 2 | 0-9:25.9.0; 1 | Logischer Name Push-Einstellungen Verbraucherinformation 2
| 3 | 1-1:1.8.1; 2 | Wirkenergie Bezug +A (QI+QIV) Tarif 1
| 4 | 1-1:1.8.2; 2 | Wirkenergie Bezug +A (QI+QIV) Tarif 2
| 5 | 1-1:2.8.1; 2 | Wirkenergie Lieferung-A (QII+QIII) Tarif 1
| 6 | 1-1:2.8.2; 2 | Wirkenergie Lieferung -A (QII+QIII) Tarif 2
| 7 | 1-1:5.8.1; 2 | Blindenergie +Ri (QI) Tarif 1
| 8 | 1-1:5.8.2; 2 | Blindenergie +Ri (QI) Tarif 2
| 9 | 1-1:6.8.1; 2 | Blindenergie +Rc (QII) Tarif 1
| 10 | 1-1:6.8.2; 2 | Blindenergie +Rc (QII) Tarif 2
| 11 | 1-1:7.8.1; 2 | Blindenergie -Ri (QIII) Tarif 1
| 12 | 1-1:7.8.2; 2 | Blindenergie -Ri (QIII) Tarif 2
| 13 | 1-1:8.8.1; 2 | Blindenergie -Rc (QIV) Tarif 1
| 14 | 1-1:8.8.2; 2 | Blindenergie -Rc (QIV) Tarif 2
### "1-hour" interval payload
_I think this is a typo and should be 15-minutes, this payload is out of interest anyway_
| # | Logischer Name; Attributindex | Register
| -- | -- | --
| 1 | 0-11:25.9.0; 2 | Objektliste Push-Einstellungen Verbraucherinformation 4
| 2 | 0-11:25.9.0; 1 | Logischer Name Push-Einstellungen Verbraucherinformation 4
| 3 | 0-1:96.1.0; 2 | M-Bus Geräte-ID 1 Kanal 1
| 4 | 0-2:96.1.0; 2 | M-Bus Geräte-ID 1 Kanal 2
| 5 | 0-3:96.1.0; 2 | M-Bus Geräte-ID 1 Kanal 3
| 6 | 0-4:96.1.0; 2 | M-Bus Geräte-ID 1 Kanal 4
| 7 | 0-1:24.2.1; 2 | M-Bus Wert 1 Kanal 1
| 8 | 0-2:24.2.1; 2 | M-Bus Wert 1 Kanal 2
| 9 | 0-3:24.2.1; 2 | M-Bus Wert 1 Kanal 3
| 10 | 0-4:24.2.1; 2 | M-Bus Wert 1 Kanal 4
## Telnet debug
### raw data using PuTTY
```
(V) LNG
(D) NOT Ready to update (internal clock 10:29:30 UTC, meter clock: 00:00:00, list type 3, est: 0)
(V) HDLC frame:
(V) 7E A0 84 CE  FF 03 13 12  8B E6 E7 00  E0 40 00 01
(V) 00 00 70 0F  00 3F 7A A6  0C 07 E9 02  1C 05 0B 1D
(V) 1E FF 80 00  00 02 0E 01  0E 02 04 12  00 28 09 06
(V) 00 08 19 09  00 FF 0F 02  12 00 00 02  04 12 00 28
(V) 09 06 00 08  19 09 00 FF  0F 01 12 00  00 02 04 12
(V) 00 01 09 06  00 00 60 01  00 FF 0F 02  12 00 00 02
[... 223 lines cut ...]
        <UInt16 Value="00E9" />
        <UInt32 Value="0000000E" />
        <UInt32 Value="00000121" />
        <UInt32 Value="0000001C" />
      </Structure>
    </DataValue>
  </NotificationBody>
</DataNotification>
```

**Hardware information:**
 - Country: Switzerland
 - Meter: Landis & Gyr E450
 - Encryption enabled: no
 - AMS reader: gPlugM, ESP32-C3.  (sorry, I did not know about this story before)
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.4.3
 - MQTT: yes
 - MQTT payload type: Home Assistant
 - HAN GPIO: GPIO7
 - HAN baud and parity: 2400 8E1
 - Temperature sensors: none
 - ENTSO-E API enabled: no

ArnieO, 2025-03-01:
Thank you for great input, much appreciated! 👍 

If I understand you correctly, you are missing Power per phase in your MQTT payloads.
You're right: The XML translation of your debug payload shows that they are indeed there:

[image]
 
So they should have been in for instance the HA payload. 
I think (but need @gskjold to confirm) they are just not implemented yet for L&G 450, maybe because this is the first L&G payload which includes those parameters.

I'll tag this as an enhancement, so that it can be implemented in upcoming firmware revision.

drgreenbaum, 2025-04-04:
Having the same issue with Kamstrup Omnipower. Works in the web UI but nothing over MQTT.

ArnieO, 2025-04-05:
> Having the same issue with Kamstrup Omnipower. Works in the web UI but nothing over MQTT.

Strange. Which MQTT payload are you using?

drgreenbaum, 2025-05-25:
Actually I am using HAN-P1. They have changed it for me last year and it started working much better. But this is still not working,..

ArnieO, 2025-05-25:
> Actually I am using HAN-P1. They have changed it for me last year and it started working much better. But this is still not working,.. but it is just the accumulated. Active power per phase is working.

That is strange. The P1 payload should contain power per phase.
Can you run a [Telnet debug](https://www.amsleser.no/module/ets_blog/blog?id_post=24) and post the output here?

drgreenbaum, 2025-05-25:
Ok something like this? 

(D) Received valid DSMR at 0 +699
(D) NOT Ready to update (internal clock 17:42:06 UTC, meter clock: 17:42:05, list type 4, est: 0)
(V) DSMR frame:
(V) 2F 4B 41 4D  35 0D 0A 0D  0A 30 2D 30  3A 31 2E 30
(V) 2E 30 28 32  35 30 35 32  35 31 38 34  32 31 35 53
(V) 29 0D 0A 31  2D 30 3A 31  2E 38 2E 30  28 30 30 30
(V) 32 38 30 37  33 2E 37 37  36 2A 6B 57  68 29 0D 0A
(V) 31 2D 30 3A  32 2E 38 2E  30 28 30 30  30 30 30 30
(V) 30 30 2E 30  30 30 2A 6B  57 68 29 0D  0A 31 2D 30
(V) 3A 33 2E 38  2E 30 28 30  30 30 30 31  37 32 34 2E
(V) 30 36 37 2A  6B 56 41 72  68 29 0D 0A  31 2D 30 3A
(V) 34 2E 38 2E  30 28 30 30  30 30 31 39  33 32 2E 31
(V) 34 35 2A 6B  56 41 72 68  29 0D 0A 31  2D 30 3A 31
(V) 2E 37 2E 30  28 30 30 30  30 2E 39 37  36 2A 6B 57
(V) 29 0D 0A 31  2D 30 3A 32  2E 37 2E 30  28 30 30 30
(V) 30 2E 30 30  30 2A 6B 57  29 0D 0A 31  2D 30 3A 33
(V) 2E 37 2E 30  28 30 30 30  30 2E 30 30  30 2A 6B 56
(V) 41 72 29 0D  0A 31 2D 30  3A 34 2E 37  2E 30 28 30
(V) 30 30 30 2E  31 31 39 2A  6B 56 41 72  29 0D 0A 31
(V) 2D 30 3A 32  31 2E 37 2E  30 28 30 30  30 30 2E 31
(V) 33 34 2A 6B  57 29 0D 0A  31 2D 30 3A  34 31 2E 37
(V) 2E 30 28 30  30 30 30 2E  32 30 39 2A  6B 57 29 0D
(V) 0A 31 2D 30  3A 36 31 2E  37 2E 30 28  30 30 30 30
(V) 2E 36 33 33  2A 6B 57 29  0D 0A 31 2D  30 3A 32 32
(V) 2E 37 2E 30  28 30 30 30  30 2E 30 30  30 2A 6B 57
(V) 29 0D 0A 31  2D 30 3A 34  32 2E 37 2E  30 28 30 30
(V) 30 30 2E 30  30 30 2A 6B  57 29 0D 0A  31 2D 30 3A
(V) 36 32 2E 37  2E 30 28 30  30 30 30 2E  30 30 30 2A
(V) 6B 57 29 0D  0A 31 2D 30  3A 32 33 2E  37 2E 30 28
(V) 30 30 30 30  2E 30 30 30  2A 6B 56 41  72 29 0D 0A
(V) 31 2D 30 3A  34 33 2E 37  2E 30 28 30  30 30 30 2E
(V) 30 30 30 2A  6B 56 41 72  29 0D 0A 31  2D 30 3A 36
(V) 33 2E 37 2E  30 28 30 30  30 30 2E 30  30 30 2A 6B
(V) 56 41 72 29  0D 0A 31 2D  30 3A 32 34  2E 37 2E 30
(V) 28 30 30 30  30 2E 31 30  32 2A 6B 56  41 72 29 0D
(V) 0A 31 2D 30  3A 34 34 2E  37 2E 30 28  30 30 30 30
(V) 2E 30 31 30  2A 6B 56 41  72 29 0D 0A  31 2D 30 3A
(V) 36 34 2E 37  2E 30 28 30  30 30 30 2E  30 30 37 2A
(V) 6B 56 41 72  29 0D 0A 31  2D 30 3A 33  32 2E 37 2E
(V) 30 28 32 33  33 2E 39 2A  56 29 0D 0A  31 2D 30 3A
(V) 35 32 2E 37  2E 30 28 32  33 34 2E 34  2A 56 29 0D
(V) 0A 31 2D 30  3A 37 32 2E  37 2E 30 28  32 33 33 2E
(V) 38 2A 56 29  0D 0A 31 2D  30 3A 33 31  2E 37 2E 30
(V) 28 30 30 30  2E 39 2A 41  29 0D 0A 31  2D 30 3A 35
(V) 31 2E 37 2E  30 28 30 30  31 2E 32 2A  41 29 0D 0A
(V) 31 2D 30 3A  37 31 2E 37  2E 30 28 30  30 32 2E 38
(V) 2A 41 29 0D  0A 21 43 36  35 39 0D 0A
(D) Received valid DSMR at 0 +699
(D) NOT Ready to update (internal clock 17:42:16 UTC, meter clock: 17:42:15, list type 4, est: 0)

drgreenbaum, 2025-05-31:
> > Having the same issue with Kamstrup Omnipower. Works in the web UI but nothing over MQTT.
> 
> I don't think it's the same issue, the active power per phase on the L&G E450 is not evaluated at all. Would it be possible to split this up into two issues? This one was being marked as an enhancement and it looks like [@drgreenbaum](https://github.com/drgreenbaum) has possibly found a bug related to Kamstrup.

Maybe a bug. In the image from @gskjold there was no accumulated power per phase either. I think it’s nice to have if you want easy tracking if your phases in home assistant energy dashboard.

ArnieO, 2025-06-01:
I have not seen any meter yet that reports accumulated power (kWh) per phase.