UtilitechAS/amsreader-firmware #939: Ams reader suddenly stopped saving data

ma10as123, 2025-03-15:
**Describe the bug**
Ams reader suddenly stopped saving data on power, ask for picture for better explanation

**Screenshots**

[image]
[image]
[image]

**Hardware information:**
 - Country: Norway
 - Meter: kamstrup omnipower single phase
 - Encryption enabled: no
 - AMS reader: Pow k+

**Relevant firmware information:**
 - Version: 2.4.3
 - MQTT: Yes
 - MQTT payload type: Home Assistant

 - HAN baud and parity: 2400 8E1
 - ENTSO-E API enabled: Yes

**Additional context**
Tried resetting, didn't help
reinstalled software with esptool-v4.8.1-win64, didn't help

ma10as123, 2025-03-17:
Yes, It was a typo :) I sent you a email now with alt the info i have

ArnieO, 2025-03-18:
> Yes, It was a typo :) I sent you a email now with alt the info i have

I close this here, as I've received and answered your email - and will assist your debugging directly.