UtilitechAS/amsreader-firmware #940: special characters not allowed in homeassistant MQTT payload

StefanTrethan, 2025-03-16:
**Describe the bug**
The Homeassistant MQTT integration will not parse non-printable characters (in the MeterId) correctly and all sensors in that topic will not work (read as unavailable). The HA logs will show that this payload can not be parsed.

b'{"lv":"","id":"SAGY\x05\xecA\x98","type":"","P":0,"P1":0,"P2":0,"P3":0,"Q":0,"PO":2572,"PO1":0,"PO2":0,"PO3":0,"QO":0,"I1":4.25,"I2":4.29,"I3":2.99,"U1":232.40,"U2":231.10,"U3":231.70,"PF":96,"PF1":0,"PF2":0,"PF3":0,"tPI1":0.000,"tPI2":0.000,"tPI3":0.000,"tPO1":0.000,"tPO2":0.000,"tPO3":0.000}'

**To Reproduce**
Needs a meter which has a meterId containing special characters. 
I have no clue why someone would even do that.

**Expected behavior**
MQTT payload should ignore special non-printable characters, or a switch to disable sending meterId for HA payload in the GUI settings.

**Screenshots**
sorry it's fixed already and I didn't take one.

**Hardware information:**
 - Country: Austria
 - Meter: Sagemcom
 - Encryption yes
 - AMS reader: ESP32
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: [e.g. 2.1.0]
 - MQTT: yes
 - MQTT payload type: Homeassistant
 - HAN GPIO: [e.g. GPIO5]
 - HAN baud and parity: [e.g. 2400 8E1]
 - Temperature sensors [e.g. 3xDS18B20]
 - ENTSO-E API enabled: [yes/no]

**Additional context**
To work around the problem I replaced the invalid meterId with "default" in the source code.
I am not a programmer, so this is only a dirty fix for myself and no real solution, but it did the trick.
Maybe someone can solve this issue properly for other people.

String AmsData::getMeterId() {
    return "defaut";    //this->meterId;
}

ArnieO, 2025-03-17:
Are you running v2.3.x firmware? 
We did some improvements on this in v2.4.0, see https://github.com/UtilitechAS/amsreader-firmware/releases/tag/v2.4.0

If you're running v2.3.x, please upgrade.

StefanTrethan, 2025-03-17:
*** Remote debug - over telnet - for ESP32 - version 3.0.5
* Host name: ams-cc IP:10.0.0.23 Mac address:24:4C:AB:82:CC:00
* Free Heap RAM: 177096
* ESP SDK version: 4.4.5.230722
******************************************************
* Commands:
    ? or help -> display these help of commands
    q -> quit (close this connection)
    m -> display memory available
    v -> set debug level to verbose
    d -> set debug level to debug
    i -> set debug level to info
    w -> set debug level to warning
    e -> set debug level to errors
    s -> set debug silence on/off
    l -> show debug level
    t -> show time (millis)
    profiler:
      p      -> show time between actual and last message (in millis)
      p min  -> show only if time is this minimal
      P time -> set debug level to profiler
    c -> show colors
    filter:
          filter <string> -> show only debugs with this
          nofilter        -> disable the filter
* Please type the command and press enter to execute.(? or h for this help)
***
(V) MBUS frame:
(V) 68 FA FA 68  53 FF 00 01  67 DB 08 53  41 47 59 05
(V) EC 41 98 81  F8 20 00 1A  96 E7 31 50  70 72 F2 5F
(V) 3E D3 4E 8F  B8 D1 43 29  C0 3A 22 8E  B6 99 2B 88
(V) 29 37 B9 59  1C 9A 24 7A  B9 FD 7A F4  CB FE 39 75
(V) EC 72 C0 8E  E9 E1 3E A9  57 86 CB 01  B4 62 DD A6
(V) 90 DC 20 2D  A9 8A 35 8A  43 09 77 B9  6A FA 8F 6A
(V) 1B FD BD A5  68 D0 9D 6A  96 0B 00 B3  5A 37 00 57
(V) 49 04 2A 44  1C AB 10 1D  B5 F5 03 4F  7F A1 D9 DF
(V) FA 74 06 67  22 89 CE E8  F6 42 D2 B3  D4 7A 4D 03
(V) 2F B6 E7 57  83 C1 EC B2  A8 8B B8 B9  9D F9 98 E8
(V) 90 91 CE 3E  8B B9 D0 C7  B8 90 0B A1  54 2E 6E 0C
(V) 42 89 E0 55  45 13 C1 74  6D C5 84 53  EB CE F2 B9
(V) 7D FD F9 69  07 22 C4 30  7F A1 FF 49  C8 BB B9 68
(V) 86 2F D9 AC  D7 51 08 B4  3F 78 E7 6A  15 ED 85 1F
(V) F3 DF FD DD  B1 5F B7 3C  70 63 51 59  BE E6 F4 F5
(V) 4C FA 19 17  01 BB 19 18  AA 98 4D 60  B7 64 49 16
(V)
(V) MBUS frame:
(V) 68 14 14 68  53 FF 11 01  67 70 54 E7  18 50 F9 4D
(V) 47 E4 06 9B  39 09 72 DC  80 16
(V) GCM frame:
(V) DB 08 53 41  47 59 05 EC  41 98 81 F8  20 00 1A 96
(V) E7 0F 80 11  7E 42 0C 07  E9 03 11 01  09 15 19 00
(V) FF C4 02 02  23 09 0C 07  E9 03 11 01  09 15 19 00
(V) FF C4 02 09  06 01 00 01  08 00 FF 06  00 41 81 E8
(V) 02 02 0F 00  16 1E 09 06  01 00 02 08  00 FF 06 00
(V) 8E 0E 6F 02  02 0F 00 16  1E 09 06 01  00 01 07 00
(V) FF 06 00 00  00 00 02 02  0F 00 16 1B  09 06 01 00
(V) 02 07 00 FF  06 00 00 06  65 02 02 0F  00 16 1B 09
(V) 06 01 00 20  07 00 FF 12  08 F7 02 02  0F FF 16 23
(V) 09 06 01 00  34 07 00 FF  12 08 FB 02  02 0F FF 16
(V) 23 09 06 01  00 48 07 00  FF 12 08 FC  02 02 0F FF
(V) 16 23 09 06  01 00 1F 07  00 FF 12 01  06 02 02 0F
(V) FE 16 21 09  06 01 00 33  07 00 FF 12  01 0B 02 02
(V) 0F FE 16 21  09 06 01 00  47 07 00 FF  12 00 ED 02
(V) 02 0F FE 16  21 09 06 01  00 0D 07 00  FF 12 03 A2
(V) 02 02 0F FD  16 FF 09 0C  31 37 38 32  31 30 33 36
(V) 39 33 36 38
(V) DLMS frame:
(V) 0F 80 11 7E  42 0C 07 E9  03 11 01 09  15 19 00 FF
(V) C4 02 02 23  09 0C 07 E9  03 11 01 09  15 19 00 FF
(V) C4 02 09 06  01 00 01 08  00 FF 06 00  41 81 E8 02
[... 232 lines cut ...]
(V) 23 09 06 01  00 1F 07 00  FF 12 01 05  02 02 0F FE
(V) 16 21 09 06  01 00 33 07  00 FF 12 01  0A 02 02 0F
(V) FE 16 21 09  06 01 00 47  07 00 FF 12  00 EB 02 02
(V) 0F FE 16 21  09 06 01 00  0D 07 00 FF  12 03 A2 02
(V) 02 0F FD 16  FF 09 0C 31  37 38 32 31  30 33 36 39
(V) 33 36 38
(D) Received valid DLMS at 35 +225
(V) Using application data:
(V) 02 23 09 0C  07 E9 03 11  01 09 15 28  00 FF C4 02
(V) 09 06 01 00  01 08 00 FF  06 00 41 81  E8 02 02 0F
(V) 00 16 1E 09  06 01 00 02  08 00 FF 06  00 8E 0E 76
(V) 02 02 0F 00  16 1E 09 06  01 00 01 07  00 FF 06 00
(V) 00 00 00 02  02 0F 00 16  1B 09 06 01  00 02 07 00
(V) FF 06 00 00  06 5C 02 02  0F 00 16 1B  09 06 01 00
(V) 20 07 00 FF  12 08 FB 02  02 0F FF 16  23 09 06 01
(V) 00 34 07 00  FF 12 08 F8  02 02 0F FF  16 23 09 06
(V) 01 00 48 07  00 FF 12 08  FC 02 02 0F  FF 16 23 09
(V) 06 01 00 1F  07 00 FF 12  01 05 02 02  0F FE 16 21
(V) 09 06 01 00  33 07 00 FF  12 01 0A 02  02 0F FE 16
(V) 21 09 06 01  00 47 07 00  FF 12 00 EB  02 02 0F FE
(V) 16 21 09 06  01 00 0D 07  00 FF 12 03  A2 02 02 0F
(V) FD 16 FF 09  0C 31 37 38  32 31 30 33  36 39 33 36
(V) 38
(V) DLMS
 (D) NOT Ready to update (internal clock 08:21:42 UTC, meter clock: 00:00:00, list type 4, est: 0)

StefanTrethan, 2025-03-17:
Regarding your earlier question, I was using 2.3.9 but I downloaded the latest version and believe the error was still present before I changed the source code.

StefanTrethan, 2025-03-17:
I'm not able to identify the ID here. It should start SAGY...
MQTT Explorer showed the same thing, SAGY and then three small rectangles for unprintable characters.

You can find some information about the meter data here (second to last page):
https://www.netz-noe.at/Download-%281%29/Smart-Meter/[218_9_SmartMeter_Kundenschnittstelle_lektoriert_14.aspx

I believe a workable fix would be to filter out those characters in the meterId string or provide more options what data to send via MQTT.

But in any case it's no big deal since anyone can get the source code, thank you for that!

StefanTrethan, 2025-03-17:
[218_13_SmartMeter_Kundenschnittstelle_2604_web.pdf](https://github.com/user-attachments/files/19284395/218_13_SmartMeter_Kundenschnittstelle_2604_web.pdf)

ArnieO, 2025-03-17:
> [218_13_SmartMeter_Kundenschnittstelle_2604_web.pdf](https://github.com/user-attachments/files/19284395/218_13_SmartMeter_Kundenschnittstelle_2604_web.pdf)

Thank you! 
The section "DLMS/COSEM Interpretation" confirms that the payload does not contain a meterID. 
So that ID apparently included by the firmware, after having identified the meter as a Sagemcom - but there is apparently a bug.

I will mark this as a bug so that my firmware colleague can take care of it for next firmware release.

ArnieO, 2025-03-17:
> Thank you!

And thank you for the notification! 👍

gskjold, 2025-03-24:
First fix on the list in v2.4.1, are you saying it is not fixed?
https://github.com/UtilitechAS/amsreader-firmware/releases/tag/v2.4.1

StefanTrethan, 2025-03-24:
It's fixed in v2.4.3, thanks.

HA still shows a warning but it doesn't stop any sensors from working.

Logger: homeassistant.components.mqtt.sensor
Source: components/mqtt/sensor.py:296
integration: MQTT (documentation, issues)
First occurred: 16 March 2025 at 12:18:22 (134532 occurrences)
Last logged: 07:21:45
Invalid state message '' from 'AMS_reader/energy'

MQTT explorer shows the following data:
AMS_reader
status = online
energy = {"tPI":4316.718,"tPO":9743.880,"tQI":0.000,"tQO":0.000,"rtc":"","t":"2025-03-24T06:26:25Z"}
power = {"lv":"","id":"SAGY A","type":"","P":0,"P1":0,"P2":0,"P3":0,"Q":0,"PO":624,"PO1":0,"PO2":0,"PO3":0,"QO":0,"I1":1.26,"I2":1.43,"I3":1.23,"U1":231.30,"U2":231.30,"U3":230.90,"PF":68,"PF1":0,"PF2":0,"PF3":0,"tPI1":0.000,"tPI2":0.000,"tPI3":0.000,"tPO1":0.000,"tPO2":0.000,"tPO3":0.000,"t":"2025-03-24T06:26:25Z"}
realtime = {"max":0.5,"peaks":[0.53,0.47,0.64],"threshold":5,"hour":{"use":0.00,"cost":0.00,"produced":0.07,"income":0.00},"day":{"use":1.52,"cost":0.00,"produced":0.17,"income":0.00},"month":{"use":27.85,"cost":0.00,"produced":445.82,"income":0.00},"t":"2025-03-24T06:26:25Z"}
state = {"id":"24:4C:AB:82:CC:00","name":"AMS_reader","up":502,"vcc":0.000,"rssi":-68,"temp":-127.00,"version":"v2.4.3","t":"2025-03-24T06:26:20Z"}
firmware = {"installed_version":"v2.4.3","latest_version":"v2.4.3","title":"amsreader firmware","release_url":"https://github.com/UtilitechAS/amsreader-firmware/releases","release_summary":"New version v2.4.3 is available","update_percentage":null}

closed 2025-03-24 by StefanTrethan