UtilitechAS/amsreader-firmware #945: Reboots create duplicate MQTT devices in Home Assistant

egiljae, 2025-03-25:
**Describe your problem**
I'm not sure if this issue is related to my setup or the AMS reader firmware, but it's been bothering me for a while.

When rebooting the AMS reader, it creates a new MQTT device in Home Assistant with a unique ID, leading to an accumulation of MQTT devices and sensors over time. I'm using the Mosquitto MQTT broker, and sensor IDs are generated in this format: "sensor.esp32_a4fd3f_". Other MQTT devices don't exhibit this behavior.

**Hardware information:**
 - Country: Norway
 - Meter: Kamstrup
 - Encryption enabled: No
 - AMS reader: Lilygo T-Internet-PoE 
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.4.5
 - MQTT: yes
 - MQTT payload type: Home-Assistant
 - HAN GPIO: GPIO4
 - HAN baud and parity: 2400 8N1
 - Temperature sensors: No
 - ENTSO-E API enabled: yes

[image]

[image]

ArnieO, 2025-03-25:
Interesting, and we've never heard of this before. 
Any idea, @gskjold ?

egiljae, 2025-03-26:
It appears to be related to Home Assistant discovery. On each reboot, a new unique item is created. Below is a snippet from the MQTT topic homeassistant/sensor.

[image]

egiljae, 2025-03-26:
I dug into the code and think I found the bug. When using Ethernet, `WiFi.getHostname()` generates the `deviceUid`, likely producing a random string. However, the correct hostname is used when on WiFi.

[image]

ArnieO, 2025-03-26:
> I dug into the code and think I found the bug.

Nice catch - thank you! We have very few users on Ethernet - thank you for helping us debug.