UtilitechAS/amsreader-firmware #948: Hours are off by 1 hour in Price configuration

tm8544, 2025-04-02:
To achieve higher price from 07:00 to 22:00, I have to set hours from 06 to 20.

So, to get this 
[image]

I have to use this hour configuration,
[image]

when the hour configuration should be like  this
[image]

I suspect that this is timezone related, my timezone is Europe/Helsinki.

Firmware is  v2.4.5

ArnieO, 2025-04-02:
Is the time indicated in your upper right corner correct?

tm8544, 2025-04-02:
Yes, the time is correct.

ArnieO, 2025-04-02:
Strange. Maybe a bug then. We'll look into it!

gskjold, 2025-04-03:
I can confirm that this is related to time zone Helsinki, will look into a fix

tm8544, 2025-05-15:
> The attached firmware should fix this issue:
>
> esp32.zip<https://github.com/user-attachments/files/20228583/esp32.zip>
> esp32c3.zip<https://github.com/user-attachments/files/20228586/esp32c3.zip>
> esp32s2.zip<https://github.com/user-attachments/files/20228582/esp32s2.zip>
> esp32s3.zip<https://github.com/user-attachments/files/20228581/esp32s3.zip>
> esp32solo.zip<https://github.com/user-attachments/files/20228584/esp32solo.zip>
> esp8266.zip<https://github.com/user-attachments/files/20228585/esp8266.zip>

Thanks, I will test it and let you know how it goes.

t: Timo

tm8544, 2025-05-15:
Addition to the previous comment: I just remembered that I did not check if hourly energy values are displayed on the right hour, or if they too are 1 hour early.
Please, check them also.

tm8544, 2025-05-22:
Now the price configuration works, there is no offset when setting hours.
And 'Future energy prices' are on correct hours, as also 'Energy use last 24 hours'.

Thanks for correcting this.