UtilitechAS/amsreader-firmware #949: E450 issue with voltage

Mysonemo, 2025-04-06:
**Describe your problem**
Hello,
I still have the issue reported here.
https://github.com/UtilitechAS/amsreader-firmware/issues/782

So volt report 24v instead of 240v. So I have enabled the multipliers to put a factor 10. But at each refresh jump to 2400v and right after to 240v. I can see that directly by staying on the GUI dashboardd of the ams reader.

**Hardware information:**
 - Country: Switzerland 
 - Meter: Landis& Gyr E450
 - Encryption enabled no
 - AMS reader:  Pow-U
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.4.5
 - MQTT: yes
 - MQTT payload type: JSON
 - HAN GPIO: [e.g. GPIO5]
 - HAN baud and parity: 2400 8E1
 - Temperature sensors 
 - ENTSO-E API enabled: [yes/no]

Thank's

Mysonemo, 2025-04-07:
Additional info.
When I disable the multiplier it stay at 24v.

[image]

ArnieO, 2025-04-07:
The described behavior is unexpected. 
Can you please clarify: Before entering voltage modifier: Is the displayed voltage stable at around 24V, or is it jumping between 24 and 240 V?

If it is jumping: Can you please run a Telnet debug over a period?
It will enable us to check if the payload consistently sends the same number of digits. 
https://www.amsleser.no/module/ets_blog/blog?id_post=24

Mysonemo, 2025-04-07:
As you can see it's It's not jumping between 24 and 240v but when i enable the multiplier it's jumping between 240v and 2400v.

[telnet.log](https://github.com/user-attachments/files/19630115/telnet.log)

ArnieO, 2025-04-07:
From what I can see, this payload gives 3 phase voltages without decimals, presented in an array at the end of the payload. 
I checked a number of payloads in the file you sent, and all of them sends 3 digits / no decimals. 

I suspect this is a new payload format we have not encountered before.
Can you please indicate which grid company this is?

I prefer we add it to our payload interpreter, and in parallel we'll investigate the reason for the jumping values you see - but I think the reason might be this new payload format.

Mysonemo, 2025-04-08:
It's a swiss provider called "Romande énergie". Thank you for the investigation ! 👍🏻

ArnieO, 2025-04-08:
Grazcha fitg!

ArnieO, 2025-04-10:
We have identified this as a bug, @gskjold will get back here and post a test version of the firmware before integrating in next upgrade.

Mysonemo, 2025-04-10:
Thanks a lot

Mysonemo, 2025-04-11:
I will be happy to test and confirm the fix :)

ArnieO, 2025-04-11:
> I will be happy to test and confirm the fix :)

Again: Grazcha fitg!
And @gskjold will post a test firmware here as soon as he gets a chance to look into this.

gskjold, 2025-04-27:
@Mysonemo please try the firmware below, use this short guide to install: https://github.com/UtilitechAS/amsreader-firmware/wiki/Flashing-via-web-GUI

[esp32s2.zip](https://github.com/user-attachments/files/19928612/esp32s2.zip)

ArnieO, 2025-05-15:
Pow-U with C3 does not exist! 
But @gskjold will provide a test firmware for C3 within shortly.

gskjold, 2025-05-15:
Attached is firmware for all chip types with this fix:

[esp32.zip](https://github.com/user-attachments/files/20226020/esp32.zip)
[esp32c3.zip](https://github.com/user-attachments/files/20226023/esp32c3.zip)
[esp32s2.zip](https://github.com/user-attachments/files/20226025/esp32s2.zip)
[esp32s3.zip](https://github.com/user-attachments/files/20226021/esp32s3.zip)
[esp32solo.zip](https://github.com/user-attachments/files/20226024/esp32solo.zip)
[esp8266.zip](https://github.com/user-attachments/files/20226022/esp8266.zip)

gskjold, 2025-05-15:
Scratch the previous build, here is a new one:

[esp32.zip](https://github.com/user-attachments/files/20227691/esp32.zip)
[esp32c3.zip](https://github.com/user-attachments/files/20227692/esp32c3.zip)
[esp32s2.zip](https://github.com/user-attachments/files/20227693/esp32s2.zip)
[esp32s3.zip](https://github.com/user-attachments/files/20227694/esp32s3.zip)
[esp32solo.zip](https://github.com/user-attachments/files/20227690/esp32solo.zip)
[esp8266.zip](https://github.com/user-attachments/files/20227695/esp8266.zip)

Mysonemo, 2025-05-18:
I confirm it works perfectly !! Great 👍🏻
Thanks a lot !!