UtilitechAS/amsreader-firmware #951: Can I rescue device bricked during update?

fa2k, 2025-05-03:
**Describe your problem**

My power bank shut off during firmware upgrade, so the device lost power. The device turns on with a blue and green light. There is no AP, and it doesn't connect to my WiFi. Buttons don't do anything.

It's a version 1.5 board so no USB flashing.

I tried to flash it using a FTDI adapter, but I get the error from esptool: "No serial data received." I get nothing on the receive light of the adapter, but there is a lot of activity on the transmit light, so it's the right COM port.

Command: `esptool.exe erase_flash`

Is there any hope for this one?

**Hardware information:**
 - AMS reader: Pow-U version 1.5

**Relevant firmware information:**
 - Version: unknown
 - Temperature sensors: no

ArnieO, 2025-05-04:
Hei Marius, 

It should work with an FTDI adapter. 
Did you connect it correctly to the board? 
The Rx/Tx marking on the Pow-U board refers to the lines on the FTDI side, meaning: 
- Rx line on FTDI should be connected to Rx
- Tx line on FTDI should be connected to Tx
- GND on FTDI must be connected to the board

And you need to power the Pow-U from USB while you do this.

fa2k, 2025-05-04:
Thanks for the reply, now I could connect and erase the flash :) As you suggested, I had to use USB power connected, and only the Rx Tx and GND pins connected, and then do the button sequence from the wiki. Can I use the command from the wiki for the version 1.5 board? I think the chip should be esp8266 instead, right? But everything else the same?

esptool --chip esp32s2 --port <port> --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 bootloader.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 firmware.bin

fa2k, 2025-05-04:
Never mind I found out there's a .sh file in the zip with the firmware. Using this command to flash the device worked 👍 It's now working well again.  Thanks a lot for the reply and tips.

ArnieO, 2025-05-04:
That's right - Pow-U layout v1.5 has an ESP8266 module. Your board should look like this:
[image]

You need to download this binary file: https://github.com/UtilitechAS/amsreader-firmware/releases/download/v2.4.5/ams2mqtt-esp8266-2.4.5.bin

And then use this flashing command: 
`esptool --chip esp8266 --port <port> --baud 115200 --before default_reset --after hard_reset write_flash 0x0 ams2mqtt-esp8266-2.4.5.bin`
Where you substitute `<port>` with your COM port (for instance COM3)

After flashing is finished: 

- Press RESET button (the LED will stay blue at this stage)
- Connect to AMS2MQTT
- Open http://192.168.4.1 in your browser
- Select board type: 
[image]
- Klick Save
- Connect to your Wi-Fi:
[image]
(I recommend changing "Hostname" to something that is easy to remember.)
- Click Save
- Enable One-click upgrade:
[image]
- Connect device to the meter
- It should now auto detect baud rate and parity. In case that fails, set it to 2400 / 8E1.

Good luck!

closed 2025-05-04 by ArnieO