UtilitechAS/amsreader-firmware #952: Incorrect parsing of DSMR timestamp

gskjold, 2025-05-15:
The code has a bug when reading OBIS 1.0.0 from DSMR, where it swaps month and day, making the timestamp wrong.

We have also discovered that there are multiple formats of timestamps coming from different meters that we should also support

gskjold, 2025-05-15:
Attached is a firmware that should fix this issue:

[esp32.zip](https://github.com/user-attachments/files/20225882/esp32.zip)
[esp32c3.zip](https://github.com/user-attachments/files/20225884/esp32c3.zip)
[esp32s2.zip](https://github.com/user-attachments/files/20225883/esp32s2.zip)
[esp32s3.zip](https://github.com/user-attachments/files/20225881/esp32s3.zip)
[esp32solo.zip](https://github.com/user-attachments/files/20225880/esp32solo.zip)
[esp8266.zip](https://github.com/user-attachments/files/20225885/esp8266.zip)

gskjold, 2025-05-15:
Scratch the previous build, here is a new one:

[esp32.zip](https://github.com/user-attachments/files/20227657/esp32.zip)
[esp32c3.zip](https://github.com/user-attachments/files/20227660/esp32c3.zip)
[esp32s2.zip](https://github.com/user-attachments/files/20227661/esp32s2.zip)
[esp32s3.zip](https://github.com/user-attachments/files/20227662/esp32s3.zip)
[esp32solo.zip](https://github.com/user-attachments/files/20227658/esp32solo.zip)
[esp8266.zip](https://github.com/user-attachments/files/20227659/esp8266.zip)

caversluis, 2025-05-29:
Fixed the issue

This code fixed the issue on my AMS-reader. The error in the log have been eliminated completely. The code has been running 

Thanks fro great support :-)