UtilitechAS/amsreader-firmware #956: AM550 Slovenia parse new DLMS values

Alex9premium, 2025-05-16:
**Describe your problem**
Hello,
thank you for this amazing software. I've finally managed to set the software.
Initially the power company set the data to be sent every 10 seconds with ASCII DMSR packet. It was 9600 7N1 but I always received Footer Checksum error. After using the telnet I found out that CRC was always 0000 values and did not change so the telegrams were refused but running them through translator they correctly returned OBIS codes and values.

Since then they changed to 1 second Telegrams using DLMS. I've read the other Slovenian issues and it seems like I get some need data which isn't getting parsed (specifically **Instantaneous active import power** per phase. They've sent me the info:

[image]

[image]

[image]

Here is the 1 second telegram:
```(V) HDLC frame:
(V) 7E A0 81 CF  02 23 13 85  E1 E6 E7 00  0F 00 02 0B 
(V) 41 0C 07 E9  05 10 05 17  2D 00 00 FF  88 80 02 12 
(V) 09 10 49 53  4B 31 30 33  30 37 39 32  32 38 33 30 
(V) 36 37 09 08  31 36 38 32  30 30 30 35  06 00 00 00 
(V) F6 06 00 00  00 00 06 00  00 00 00 06  00 00 01 1E 
(V) 12 09 15 12  09 11 12 09  18 12 00 98  12 00 0B 12 
(V) 00 10 06 00  00 00 D8 06  00 00 00 13  06 00 00 00 
(V) 0A 06 00 00  00 00 06 00  00 00 00 06  00 00 00 00 
(V) 56 35 7E 
(V) LLC frame:
(V) E6 E7 00 0F  00 02 0B 41  0C 07 E9 05  10 05 17 2D 
(V) 00 00 FF 88  80 02 12 09  10 49 53 4B  31 30 33 30 
(V) 37 39 32 32  38 33 30 36  37 09 08 31  36 38 32 30 
(V) 30 30 35 06  00 00 00 F6  06 00 00 00  00 06 00 00 
(V) 00 00 06 00  00 01 1E 12  09 15 12 09  11 12 09 18 
(V) 12 00 98 12  00 0B 12 00  10 06 00 00  00 D8 06 00 
(V) 00 00 13 06  00 00 00 0A  06 00 00 00  00 06 00 00 
(V) 00 00 06 00  00 00 00 
(V) DLMS frame:
(V) 0F 00 02 0B  41 0C 07 E9  05 10 05 17  2D 00 00 FF 
(V) 88 80 02 12  09 10 49 53  4B 31 30 33  30 37 39 32 
(V) 32 38 33 30  36 37 09 08  31 36 38 32  30 30 30 35 
(V) 06 00 00 00  F6 06 00 00  00 00 06 00  00 00 00 06 
(V) 00 00 01 1E  12 09 15 12  09 11 12 09  18 12 00 98 
(V) 12 00 0B 12  00 10 06 00  00 00 D8 06  00 00 00 13 
(V) 06 00 00 00  0A 06 00 00  00 00 06 00  00 00 00 06 
(V) 00 00 00 00  
(D) Received valid DLMS at 30 +98
(V) Using application data:
(V) 02 12 09 10  49 53 4B 31  30 33 30 37  39 32 32 38 
(V) 33 30 36 37  09 08 31 36  38 32 30 30  30 35 06 00 
(V) 00 00 F6 06  00 00 00 00  06 00 00 00  00 06 00 00 
(V) 01 1E 12 09  15 12 09 11  12 09 18 12  00 98 12 00 
(V) 0B 12 00 10  06 00 00 00  D8 06 00 00  00 13 06 00 
(V) 00 00 0A 06  00 00 00 00  06 00 00 00  00 06 00 00 
(V) 00 00 
(V) DLMS
```
Translated:
```
<DataNotification>
  <LongInvokeIdAndPriority Value="00020B41" />
  <!--2025-05-16 23:45:00-->
  <DateTime Value="07E9051005172D0000FF8880" />
  <NotificationBody>
    <DataValue>
      <Structure Qty="12" >
        <!--ISK1030792283067-->
        <OctetString Value="49534B31303330373932323833303637" />
        <!--16820005-->
        <OctetString Value="3136383230303035" />
        <UInt32 Value="000000F6" />
        <UInt32 Value="00000000" />
        <UInt32 Value="00000000" />
        <UInt32 Value="0000011E" />
        <UInt16 Value="0915" />
        <UInt16 Value="0911" />
        <UInt16 Value="0918" />
        <UInt16 Value="0098" />
        <UInt16 Value="000B" />
        <UInt16 Value="0010" />
        <UInt32 Value="000000D8" />
        <UInt32 Value="00000013" />
        <UInt32 Value="0000000A" />
        <UInt32 Value="00000000" />
        <UInt32 Value="00000000" />
        <UInt32 Value="00000000" />
      </Structure>
    </DataValue>
  </NotificationBody>
</DataNotification>
```
15 min interval:
```
(V) HDLC frame:
(V) 7E A0 6B CF  02 23 13 CB  86 E6 E7 00  0F 00 02 0B 
(V) 40 0C 07 E9  05 10 05 17  2D 00 00 FF  88 80 02 0C 
(V) 09 10 49 53  4B 31 30 33  30 37 39 32  32 38 33 30 
(V) 36 37 09 08  31 36 38 32  30 30 30 36  16 01 09 02 
(V) 00 02 06 00  07 31 FB 06  00 03 34 F3  06 00 03 FD 
(V) 08 06 00 00  AE 4C 06 00  00 6F 77 06  00 00 3E D5 
(V) 06 00 00 50  1B 06 00 04  C9 ED 33 2D  7E 
(V) LLC frame:
(V) E6 E7 00 0F  00 02 0B 40  0C 07 E9 05  10 05 17 2D 
(V) 00 00 FF 88  80 02 0C 09  10 49 53 4B  31 30 33 30 
(V) 37 39 32 32  38 33 30 36  37 09 08 31  36 38 32 30 
(V) 30 30 36 16  01 09 02 00  02 06 00 07  31 FB 06 00 
(V) 03 34 F3 06  00 03 FD 08  06 00 00 AE  4C 06 00 00 
(V) 6F 77 06 00  00 3E D5 06  00 00 50 1B  06 00 04 C9 
(V) ED 
(V) DLMS frame:
(V) 0F 00 02 0B  40 0C 07 E9  05 10 05 17  2D 00 00 FF 
(V) 88 80 02 0C  09 10 49 53  4B 31 30 33  30 37 39 32 
(V) 32 38 33 30  36 37 09 08  31 36 38 32  30 30 30 36 
(V) 16 01 09 02  00 02 06 00  07 31 FB 06  00 03 34 F3 
(V) 06 00 03 FD  08 06 00 00  AE 4C 06 00  00 6F 77 06 
(V) 00 00 3E D5  06 00 00 50  1B 06 00 04  C9 ED 
(D) Received valid DLMS at 30 +76
(V) Using application data:
(V) 02 0C 09 10  49 53 4B 31  30 33 30 37  39 32 32 38 
(V) 33 30 36 37  09 08 31 36  38 32 30 30  30 36 16 01 
(V) 09 02 00 02  06 00 07 31  FB 06 00 03  34 F3 06 00 
(V) 03 FD 08 06  00 00 AE 4C  06 00 00 6F  77 06 00 00 
(V) 3E D5 06 00  00 50 1B 06  00 04 C9 ED  
(V) DLMS
```
Translated:
```
<DataNotification>
  <LongInvokeIdAndPriority Value="00020B40" />
  <!--2025-05-16 23:45:00-->
  <DateTime Value="07E9051005172D0000FF8880" />
  <NotificationBody>
    <DataValue>
      <Structure Qty="0C" >
        <!--ISK1030792283067-->
        <OctetString Value="49534B31303330373932323833303637" />
        <!--16820006-->
        <OctetString Value="3136383230303036" />
        <Enum Value="01" />
        <OctetString Value="0002" />
        <UInt32 Value="000731FB" />
        <UInt32 Value="000334F3" />
        <UInt32 Value="0003FD08" />
        <UInt32 Value="0000AE4C" />
        <UInt32 Value="00006F77" />
        <UInt32 Value="00003ED5" />
        <UInt32 Value="0000501B" />
        <UInt32 Value="0004C9ED" />
      </Structure>
    </DataValue>
  </NotificationBody>
</DataNotification>
```

**Hardware information:**
 - Country: Slovenia
 - Provider: Elektro Ljubljana
 - Meter: Iskraemeco AM550
 - Encryption enabled no
 - AMS reader: ESP32C3 Supermini
 - M-bus adapter (if applicable): P1

**Relevant firmware information:**
 - Version: 2.4.5
 - MQTT: yes
 - MQTT payload type: Home Assistant
 - HAN GPIO: GPIO2
 - HAN baud and parity: 38400 8N1

ArnieO, 2025-05-19:
Hi Alex, 

Thank you for your very well-structured input! 
However, I am a bit confused - as my impression is clearly that we are already parsing correctly the data from Elektro Ljubljana
/ Iskraemeco AM550. We have several users in your area - both on home made hardware and on units bought from us. And since our last upgrade of the parsing of Elektro Ljubljana payloads (in January), things seem to work correctly.

Can you be so kind as to read through this thread, and see if you find information there that can help you?
https://github.com/UtilitechAS/amsreader-firmware/issues/869

Alex9premium, 2025-05-19:
Hi,

so after reading the thread I see that my telegram payload is larger. Everything is parsed except for Phase Power (wattage).

[image]

Comparing it do the structure of the telegram I've been sent it seems it is being reported:
```
<DataNotification>
  <LongInvokeIdAndPriority Value="00020B41" />
  <!--2025-05-16 23:45:00-->
  <DateTime Value="07E9051005172D0000FF8880" />
  <NotificationBody>
    <DataValue>
      <Structure Qty="12" >
        <!--ISK1030792283067-->
        <OctetString Value="49534B31303330373932323833303637" />
        <!--16820005-->
        <OctetString Value="3136383230303035" />
        <UInt32 Value="000000F6" />
        <UInt32 Value="00000000" />
        <UInt32 Value="00000000" />
        <UInt32 Value="0000011E" />
        <UInt16 Value="0915" />
        <UInt16 Value="0911" />
        <UInt16 Value="0918" />
        <UInt16 Value="0098" />
        <UInt16 Value="000B" />
        <UInt16 Value="0010" />
        <UInt32 Value="000000D8" />   <- Import power L1
        <UInt32 Value="00000013" />    <- Import power L2
        <UInt32 Value="0000000A" />   <- Import power L3
        <UInt32 Value="00000000" />   <- Export power L1
        <UInt32 Value="00000000" />    <- Export power L2
        <UInt32 Value="00000000" />   <- Export power L3
      </Structure>
    </DataValue>
  </NotificationBody>
</DataNotification>
```

ArnieO, 2025-05-19:
> Hi,
> 
> so after reading the thread I see that my telegram payload is larger. Everything is parsed except for Phase Power (wattage).

OK - that is interesting: There seems to be different payload versions used by Elektro Ljubljana. 
We'll include this in next firmware revision then.

ArnieO, 2025-05-22:
Hi Alex, 

We've looked through the code - and this format should have been supported (correctly parsed) - so there must be a bug here. 
We will run code testing with your data and fix this.

gskjold, 2025-05-28:
After some testing, I have found the issue, here is a test firmware:

[esp32.zip](https://github.com/user-attachments/files/20480605/esp32.zip)
[esp32c3.zip](https://github.com/user-attachments/files/20480600/esp32c3.zip)
[esp32s2.zip](https://github.com/user-attachments/files/20480602/esp32s2.zip)
[esp32s3.zip](https://github.com/user-attachments/files/20480603/esp32s3.zip)
[esp32solo.zip](https://github.com/user-attachments/files/20480604/esp32solo.zip)
[esp8266.zip](https://github.com/user-attachments/files/20480601/esp8266.zip)

closed 2025-06-05 by gskjold