UtilitechAS/amsreader-firmware #957: Does the MQTT implementation support ipv6

Nierhoff, 2025-05-19:
**Describe your problem**
I am wondering if amsreader mqtt implementation is supporting ipv6?

My MQTT server is only accessible on ipv6. I can connect to it using a mqtt explore on the ipv6 address from my laptop, but in the amsreader UI i only see "MQTT: Connection failed", my laptop and the amsreader is on the same network.

By telnet'ing to the hardware i can see this error:
"(W) MQTT connection not available, double blink"
But that is not saying much more then the UI.

The Device both have a ipv6 address and a ipv6 link local address, and the amsreader itself is accessible via its ipv6 address. 

**Hardware information:**
 - Country: Denmark
 - Meter: Kamstrup
 - Encryption enabled [yes/no]
 - AMS reader: Pow-K+
 - M-bus adapter (if applicable):
 - Chip: esp32s2 (160MHz)

**Relevant firmware information:**
 - Version: 2.4.5
 - MQTT: yes
 - MQTT payload type: HA
 - HAN GPIO: [e.g. GPIO5]
 - HAN baud and parity: [e.g. 2400 8E1]
 - Temperature sensors [e.g. 3xDS18B20]
 - ENTSO-E API enabled: [yes/no]

ArnieO, 2025-05-19:
Good question, which I need @gskjold to answer.

ArnieO, 2025-05-22:
It could be that the MQTT library we're using does not support IPv6. We have to dig into this, do some testing and get back on this. 

We know that IPV6 is becoming common in Denmark - so this is an issue wee need to solve.

Nierhoff, 2025-05-22:
Thanks for update. I have temporarily solved my issue by adding IPv4 DNS (A record) name to my mqtt broker. 
I think it is a bit of an overstatement to say that IPv6 is becoming common in DK, i have only found one mobil 5G provider that supports it. 
But i agree that this issue should be solved :)