UtilitechAS/amsreader-firmware #970: Kamstrup reports very high voltage

Spexdallas, 2025-06-01:
Kamstrup sends very high voltage in gui.
Multipliers is on and shows W 1, V 1, A 1, kWh 0.1

**Hardware information:**
 - Country: Sweden
 - Meter: Kamstrup 6841138BN245101090
 - Encryption enabled No
 - AMS reader: Pow-K+
 
**Relevant firmware information:**
 - Version: v2.4.5
 - MQTT: Yes
 - MQTT payload type: JSON
 - HAN GPIO: [e.g. GPIO5]
 - HAN baud and parity: 2400 8N1
 - ENTSO-E API enabled: Yes

[image]

Spexdallas, 2025-06-01:
I have tried to roll back to 2.4.4 but same result.

ArnieO, 2025-06-01:
Interesting! 
It looks like your Kamstrup is sending voltage with two decimals, I cannot recall having seen that before. 

We know modifications have been done in Sweden (see this article: https://www.amsleser.no/module/ets_blog/blog?id_post=35 ) to increase parameter precision - but have not seen that yet on voltage. Unfortunately, there is no information in the payload on the number of decimals.

This is not in line with the HAN-NVE specification in [Kamstrup documentation](https://github.com/UtilitechAS/amsreader-firmware/blob/main/doc/Norway/Kamstrup%20HAN-NVE%20interface%20description_rev_3_1.pdf), stating there should be no scaling on the voltage (paragraph 2.7.1):

[image]

Could the reason be that you have set a modifier in your configuration? 
If not, you'll have to correct this by using a multiplier in your config:

[image]

Spexdallas, 2025-06-01:
The multiplier as you shown did the trick, thanx

ArnieO, 2025-06-02:
> The multiplier as you shown did the trick, thanx It have worked perfectly before, but now for some reason it was way out.

That is good to hear!

> It have worked perfectly before, but now for some reason it was way out.

This means there must have been a change in your payload recently - meaning your meter must have received a firmware update.
And if it happened to you, many other Swedish customers might be experiencing the same.

Can I ask which grid company (nätbolag) you are attached to?

ArnieO, 2025-06-05:
Bump: Which grid company (nätbolag) is this?

ArnieO, 2025-06-05:
> Sorry for late answer, its Varbergs Energi.

Thank you! I will contact them in an attempt to understand what is going on: Why did they introduce the change, and is it something that will be rolled out also in other grid companies?

In the definition of the so called OBIS codes (sent as part of the payload), it is possible to indicate the number of decimals per parameter. Sadly, no smart meter we've seen so far includes that information. If they could introduce that, it would be a significant improvement.

ArnieO, 2025-09-09:
> Same issue with my meter in Denmark (Bornholm)

Very interesting, thank you for notifying us. 
We'll update [this article](https://www.amsleser.no/module/ets_blog/blog?id_post=35) to indicate:
- Not isolated to Sweden (at least also Denmark)
- It could be 10x or 100x error
- It does not only affect power readout (as indicated in the article), other parameters may be affected. 

We've been in dialogue with Kamstrup on this, and they confirm they are in a process of considering (and testing out!) different payload formats. We strongly recommended them to add information in the payload that gives information on data resolution, so that this can be handled by the firmware.

@sfinne : As this must be some updated format, we would very much like to study the decoded raw data. 
Could you be so kind as to run a [Telnet debug](https://www.amsleser.no/module/ets_blog/blog?id_post=24), and email the logfile to post@amsleser.no? Thank you in advance!

ArnieO, 2025-09-09:
Thank you @sfinne for sending a Telnet debug log, which I have analyzed and responded by email. 
The conclusion is:
- The payload is identical to what we're used to see, except there are two more digits in the voltage readings. 
- There is still no information in the payload to identify it. 
- The payload version is named "Kamstrup_V0001", which is the identifier used on all payload versions, so it is useless. 

**Conclusion**
The firmware cannot detect this, users have to do the manual correction using Multipliers in the config.