UtilitechAS/amsreader-firmware #975: Upgrade to v2.4.7 FAILD :-(

sm1oii, 2025-06-05:
My upgrade to v2.4.7 stopped at 85,8%, is my unit bricked now? What's next step?

[image]

/Harri

sm1oii, 2025-06-06:
The AMS reader worked perfectly for years, no problem with earlier upgrades but this one halted at 85,8%. Still after 9 hours the upgrade is at 85,8%. Upgrade was from v2.4.5 to v2.4.7. 

[image]

[image]

You are right, it seems not to be connected to WiFi. The WiFi signal strength was always good (green) before but this
upgrade seems to have killed it. Is my AMS reader bricked now, what can I do to get in online again?

/Harri

sm1oii, 2025-06-06:
No, the user interface is not responsive because the pow-p1 is not connected to WiFi (ping 100% loss).
I will try the USB flash.

/Harri

ArnieO, 2025-06-06:
> My upgrade from 2.4.5 to 2.4.7 (web UI) was performed without issues. Not that it helps SM1OII but just to point out that the 2.4 updates can be applied without issues. Prior to 2.4.x my updates had to be performed via USB but v. 2.4.x fixed this and the web UI update process has been rock solid ever since.

We too have a clear impression that the new upgrade method in v2.4.x in general runs very smoothly, this is the first feedback we have on a failure during upgrade. 

But for some reason, something happened for @sm1oii during the process - which is a bit disturbing. 

> No, the user interface is not responsive because the pow-p1 is not connected to WiFi (ping 100% loss).

🤦 Of course... sorry! 
The USB flashing method should get you up and running: https://wiki.amsleser.no/en/Update-firmware-usb#usb-upgrade
Let me know how it works for you!

sm1oii, 2025-06-06:
OK, now my pow-p1 is online again. The USB flash worked fine.

[image]

[image]

And my config and history are saved. Thank you for advice!

PS: This was the first time I tried to made an Web UI upgrade using Edge on my laptop. Usually I run Chrome
but for some reason AMS reader don't work on my Win 10 laptop using Chrome. All the graphs are scrolled
down so you can't see them. Works great on any other computer using Chrome. I suppose it has something
to do with Chrome on my laptop. Edge on the laptop shows all the graphs ok.

/Harri

ArnieO, 2025-06-06:
Great!

Your issue with Chrome on your laptop sounds strange, we've never heard about such issue. 
Have you tried deleting cookies?

closed 2025-06-06 by ArnieO