UtilitechAS/amsreader-firmware #978: Home assistant error: The unit of measurement C is not valid together with device class temperature

niklerus, 2025-06-15:
**Describe the bug**
HA gives a warning about wrong unit for temperature:
"The unit of measurement C is not valid together with device class temperature, this will stop working in HA Core 2025.7.0"

**To Reproduce**
Steps to reproduce the behavior:
Run AMS reader v2.4.7 on ESP32 chip and mqtt.sensor logs in HA

**Expected behavior**
Valid temperature unit sent from AMS

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware information:**
 - Country: Sweden
 - Meter: Sagemcom T211
 - Encryption enabled no
 - AMS reader: ESP32
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.4.7
 - MQTT: yes
 - MQTT payload type: home-assistant
 - HAN GPIO: GPIO3
 - HAN baud and parity: 115200 8N1
 - Temperature sensors: none
 - ENTSO-E API enabled: no

**Additional context**
I have no temperature sensor installed.
Seems HA is expecting "°C" instead of "C".

gskjold, 2025-06-19:
This is weird, because °C is what we have in the code:
https://github.com/UtilitechAS/amsreader-firmware/blob/bcb3c3b2ec04959a1adff47f8161e0a3c80fa089/lib/HomeAssistantMqttHandler/include/HomeAssistantStatic.h#L126

But maybe there is something weird going on when you don't have sensors, I will look into it

gskjold, 2025-10-01:
Unable to reproduce

niklerus, 2025-10-02:
OK, I managed to get rid of the error message by entering a numeric value in the Temperature input box.