UtilitechAS/amsreader-firmware #985: My p1-reader just died

sm1oii, 2025-07-19:
My p1-reader just disappeared from the WiFi network, it should have IP 192.168.1.138

[image]

[image]

I tried to disconnect the cable in the p1-connector for a few seconds to reset the p1-reader but that did not work, P1-reader is still not in the WiFi network. 

Do I need to flash the p1-reader again? I did that a couple of months ago after a software upgrade failure.

/Harri

ArnieO, 2025-07-19:
Try the Emergency Factory Reset procedure: https://wiki.amsleser.no/en/firmware#emergency-factory-reset

sm1oii, 2025-07-19:
Thank's, I will try it later.

/Harri

sm1oii, 2025-07-19:
I opened the box where the p1-reader is mounted , the p1-redader is flashing 3 x red and 1 x green.
What does that mean?

/Harri

sm1oii, 2025-07-19:
Now pow-p1 is online again, lost all settings. Thank´s for helping.

[image]

/Harri

closed 2025-09-04 by gskjold