UtilitechAS/amsreader-firmware #986: Kaifa MA309M encrypted with Slimmelezer+ HAN Serial buffer full

mwasowski, 2025-07-27:
**Describe your problem**
I was able to setup the hardware part, as far I was able to tell Slimmelezer+ RX is on D7 which is GPIO13. I am not getting RX error. Instead I get `Serial buffer full`. In guides for other meters I saw that buffer should be set to 512 but I can only set it to 256. I don't know if it's limitation of Wemos D1 Mini and ESP8266 or it's a different issue. 

**Hardware information:**
 - Country: Poland
 - Meter: Kaifa MA309A
 - Encryption enabled: yes
 - AMS reader: Slimmelezer+
 - M-bus adapter (if applicable): ni

**Relevant firmware information:**
 - Version: 2.4.7
 - MQTT: no
 - MQTT payload type: no
 - HAN GPIO: GPIO13
 - HAN baud and parity: 2400 8N1
 - Temperature sensors: no
 - ENTSO-E API enabled: no

mwasowski, 2025-07-28:
Hi @ArnieO, that is understandable.

The operator is Stoen Operator.
With debug enabled and 2400 baud I get:

<details>
<summary>Log</summary>

```
(E) Serial buffer overflow
(E) Serial buffer overflow
(E) Serial buffer overflow
(E) Serial buffer overflow
(E) Serial buffer overflow
(E) Serial buffer overflow
(E) Serial buffer overflow
(E) Serial buffer overflow
(E) Serial buffer overflow
(E) Serial buffer overflow
(E) Serial buffer overflow
(E) Serial buffer overflow
(E) Serial buffer overflow
(E) Serial buffer overflow
(E) Serial buffer overflow
(E) Serial buffer overflow
(E) Serial buffer overflow
(E) Serial buffer overflow
(E) Serial buffer overflow
(E) Serial buffer overflow
(E) Serial buffer overflow
(E) Serial buffer overflow
(E) Serial buffer overflow
(E) Serial buffer overflow
(E) Serial buffer overflow
(E) Serial buffer overflow
(E) Serial buffer overflow
(E) Serial buffer overflow
(E) Serial buffer overflow
(E) Serial buffer overflow
(E) Serial buffer overflow
(W) No HAN data received last 30s, single blink
```

Then I switched to 38400 and started getting payload:
```
(I) (setupHanPort) Setting up HAN on pin 13/-1 with baud 38400 and parity 3
(D) Software serial
(D) Using serial buffer size 128
(E) Ended up in default case while unwrapping...(tag AB)
(W) Unknown data received
(V)   payload:
(V) AB 2A 96 2C  11 44 65 B5  F7 A9 7B 56  25 4B 5D 95
(V) 56 DD 9F 2D  F5 AD 77 5B  F7 75 F1 49  56 BA 33 8B
(V) FB F5 4A 21  CD FD C9 9C  D5 B4 B5 4B  57 33 3A AE
(V) D1 CD AE E8  72 ED 66 6A  6F B4 6F 6C  6C A5 5D 87
(V) 99 44 59 DA  55 F5 EE D7  49 AE A6 B6  9F D9 D4 7B
(V) 52 2A 37 6C  55 5D AE 7B  B5 DB C5 4F  7B E2 E9 AB
(V) DF 5D 7A 4A  6A 3B D7 7F  34 57 D1 2B  C6 59 56 74
(V) DD 49 FD E5  75 B9 3F 57  3A EB 6F 5B  B5 55 F6 57
(V) 6B 52 6A AB  F9
(E) Serial buffer overflow
(I) Increasing RX buffer to 384 bytes
(W) No HAN data received last 30s, single blink
(E) Ended up in default case while unwrapping...(tag AB)
(W) Unknown data received
(V)   payload:
(V) AB 2A 96 62  44 65 6D FB  AA 75 75 BB  22 2D E9 AB
(V) 6B BB DD AD  5C 75 A6 5A  B2 AF A3 3F  47 4A E5 7D
(V) FF 5A 71 6B  4D 57 AE 7E  7F 5E AD 97  57 9F DB D7
(V) B1 74 DB 23  59 B5 B8 4A  13 2D A7 4F  95 7E 57 5B
[... 327 lines cut ...]
(V) 1F 7B 12 FF  76 FF FE FF  A8 FF 74 A0  86 FE B1 AF
(V) 01 46 FF D2  E1 FD 93 FF  C0 63 22 BB  F9 D6 81 FF
(V) 9A E4 FB 00  FE
(V)  - Failed! (-2)
(V) DSMR wants to decrypt at position 20, length: 149, payload:
(V) 00 82 02 8F  30 00 00 00  00 45 1A 30  A1 38 28 55
(V) 7A 57 8E 74  DA 6E 51 CE  47 01 BD EA  C6 DB CC 9F
(V) 30 F4 FB 8B  FF DC 82 38  FA 42 F3 CB  31 FE 2C E0
(V) 9C 48 FF 6F  D3 1F 78 32  F9 91 CE FA  98 9A 8D FE
(V) E8 1D BA 60  82 82 D2 33  FF FE 1B 9D  CC E5 46 67
(V) FF 3C FE A1  F5 C7 F9 0A  66 FF 4D EC  3F F3 28 FF
(V) C9 B2 F7 DE  EF 98 FD FE  FF FF FE 31  02 FF 28 FF
(V) 1F 7B 12 FF  76 FF FE FF  A8 FF 74 A0  86 FE B1 AF
(V) 01 46 FF D2  E1 FD 93 FF  C0 63 22 BB  F9 D6 81 FF
(V) 9A E4 FB 00  FE
```
</details>

Does that make any sense to you? 🤔 

I'm attaching log, original HAN specification from operator page ([link](https://www.stoen.pl/strona/liczniki-zdalnego-odczytu)) and rough translation of it.

[telnet.log](https://github.com/user-attachments/files/21465687/telnet.log)
[interfejs-sieci-han-w-stoen-operator-v1-0-1.pdf](https://github.com/user-attachments/files/21465688/interfejs-sieci-han-w-stoen-operator-v1-0-1.pdf)
[HAN_Interface_Stoen_Operator_EN.pdf](https://github.com/user-attachments/files/21465689/HAN_Interface_Stoen_Operator_EN.pdf)

ArnieO, 2025-07-28:
The non-inverted version looked better. 

The payload from a P1 meter is quite large, and it could well be that the problem here is the maximum buffer size on ESP8266 when running on a P1 meter (and especially with encryption). We have crammed a maximum of functionality into our firmware - and switched from ESP8266 to ESP32 a long time ago for our own products (see https://amsleser.no) in order to have enough resources.

As you're not running this on our own product, our main interest will be to confirm that we can read the payload from a Polish Kaifa MA309A meter - which is a new model for us. It should in principle be fine, if the meter follows the [DSMR P1 Companion Standard](https://www.netbeheernederland.nl/sites/default/files/2024-02/dsmr_5.0.2_p1_companion_standard.pdf), but we've seen "creative" solutions before, so it would be great to have it tested.

It could be that we'll provide you with a stripped-down test version of the firmware, with some functionality removed in order to be able to increase the buffer size. 

Anyway, we'll have to wait for my firmware colleague to be back from vacation > End August.

ArnieO, 2025-07-29:
> As for the standard, I have following documents:
> 
> [model-obiektow-dla-licznikow-bezposrednich-plc-v2.0.pdf](https://github.com/user-attachments/files/21470584/model-obiektow-dla-licznikow-bezposrednich-plc-v2.0.pdf) [opis-dziennikow-zdarzen-oraz-lista-kodow-zdarzen-v1-0.pdf](https://github.com/user-attachments/files/21470586/opis-dziennikow-zdarzen-oraz-lista-kodow-zdarzen-v1-0.pdf) [opis-profili-rejestrowane-kody-obis-v1-0-2.pdf](https://github.com/user-attachments/files/21470628/opis-profili-rejestrowane-kody-obis-v1-0-2.pdf)
> 
> Though I don't know if those will help. If you need translation, let me know :)

This could potentially be useful, and may not even need translation (we'll ask if we need assistance). Thank you!

> All this is actually because I'm a dumbass and didn't read that Slimmelezer is a DSMR device and apparently meters my grid operator installed use DLMS COSEM communication 🤦‍♂️ So I'm trying to salvage the situation. Sorry for the trouble..

We have had our first sales to Poland this year - and are looking forward to many more! 😉 

We were recently through a process with the Austrian energy company Netz Burgenland, that also uses encrypted P1 meters (Landis & Gyr E360). I'm not fully familiar with the technical details, but it was apparently the encryption that made it necessary to use DSMR in stead of DLMS COSEM - and there were alternative ways to configure the HAN port of that meter. Before starting rolling them out, they contacted us to discuss which configuration to use so that they were sure there was an adapter on the market that could read it. They bought Pow-P1 readers from us for testing, and sent us the decrypted Telnet debug raw data for different meter configuration settings.

The result of their investigation is that our [Pow-P1](https://www.amsleser.no/home/10-39-pow-p1-reader-for-power-meter-with-p1-port.html#/23-enclosure-with/27-cable-rj12_30_cm) is now listed on [their website](https://www.netzburgenland.at/smart-meter/) as compatible with their meter:

[image]

mwasowski, 2025-07-29:
Well, I doubt Stoen Operator will go that route. They just say "go find a device online, we're not giving any recommendations from out side" :/ I already asked when requesting for the remote access feature to be enabled :)

Nice to hear you had some sales to Poland! I think there are a log of Gama 150 and 350 devices here. Maybe other grid providers are more willing to support, dunno :/

I'll try to look into either using the device I have or trying to frankenstein something with ESP32 I probably have lying around..

And ofcourse I have the option to purchase Pow-P1, but it damn expensive actually :O

ArnieO, 2025-07-29:
> I'll try to look into either using the device I have or trying to frankenstein something with ESP32 I probably have lying around..

In that case you need to study the physical P1 interface in order to get data from the meter. And note that the data line is an Open Collector/drain output - so you must pull it to 3,3V on your side with a suitable resistor.

mwasowski, 2025-08-05:
Hi @ArnieO, sorry for the delay. Here is the photo of my unit,

[image]

ArnieO, 2025-08-05:
> Hi [@ArnieO](https://github.com/ArnieO), sorry for the delay. Here is the photo of my unit,

Thank you very much! 

So the model number is MA309M (the thread title is correct, but your first post in the thread says MA309A).
And the right-hand side lid for the cable says "P1 Port"

[image]

So this meter seems indeed to have a P1 interface according to [DSMR P1 Companion Standard](https://www.netbeheernederland.nl/sites/default/files/2024-02/dsmr_5.0.2_p1_companion_standard.pdf). Thank you for helping with the clarification!

And we'll get back regarding your decoding problems around end August.

mwasowski, 2025-08-05:
You are correct, I made a mistake providing the model number.

ArnieO, 2025-08-05:
> You are correct, I made a mistake providing the model number.

No problem, thanks for helping to sort it out.

mwasowski, 2025-08-05:
> No problem, thanks for helping to sort it out. 
Updated the original message.

mwasowski, 2025-08-30:
Hi, with encryption turned off I get:
```
(V) CRC incorrrect, 0000 != 85B1 at position 428
(V) DSMR frame:
(V) 2F 4B 46 4D  35 4B 41 49  46 41 2D 4D  45 54 45 52
(V) 0D 0A 0D 0A  00 82 02 8F  30 00 00 00  00 45 1A 30
(V) A1 38 28 55  7A 57 8E 74  DA 6E 5E FF  82 D2 33 FF
(V) FE FE 1B 9D  CC E5 46 FF  67 FF 95 39  FF A6 58 FC
(V) C1 F9 0A 66  F9 FC 3F F3  17 C9 DE B2  FE 2A DE E2
(V) 4D 96 FD E9  39 FF C2 FC  5E 29 C3 98  CD 02 C1 FA
(V) 48 FE FD 8A  FE 31 02 28  1F 7C 3C E3  BF 7B FF 43
(V) FF F5 F7 EE  94 FF 23 9E  76 27 FC D7  76 E0 1C C8
(V) 79 9D 00 EB  01 3B 46 D2  FF FD 93 FF  1C C0 E3 22
(V) BB E9 42 FF  FF CD D6 FF  81 EF 9A 5F  1B FF B0 00
(V) C6 5C 90 99  53 F9 91 C5  CB FE CF 67  70 FD D8 DA
(V) B7 E0 EE FB  AE FF AE B1  FD 06 EE 80  13 80 87 A5
(V) 62 90 FC 7C  FA 5E 3F D2  C5 F8 E4 E9  61 B1 80 A8
(V) 5E 9C 8F BD  08 72 FF C0  06 B0 06 02  38 17 FF 30
(V) E4 86 D3 60  76 90 FE 99  F9 87 4E 67  FF 3C 84 FD
(V) 83 FD F5 F9  93 FF F0 36  97 80 B5 17  63 FF 28 FF
(V) 04 C9 6C DB  FC 98 FF 2F  4B 46 4D 35  4B 41 49 46
(V) 41 2D 4D 45  54 45 52 0D  0A 0D 0A 00  82 02 8F 30
(V) 00 00 00 00  45 1A 30 A1  38 28 55 7A  57 8E 74 DA
(V) 6E 5E FF 1B  9D CC FD 66  FE E3 B3 FF  94 F9 FB 77
(V) F8 CC C1 F9  0A B8 66 89  18 3F D3 F8  75 5E 16 D1
(V) A4 DE E2 FD  96 2C FC 24  2D 39 5A C4  FE 32 FF 1F
(V) FF EF CD FE  B3 FE EF FA  48 E6 FD FA  8A FE 31 25
(V) 7C FC 23 FF  C6 FB F7 43  DD FF CA 7B  12 23 76 FF
(V) E4 F7 FE 76  E0 1C C8 79  9D 00 AA E7  E8 C5 AA FF
(V) E8 A0 FF 1C  FF 38 C0 63  E2 FA BB 17  F9 CF F6 B5
(V) 81 AF 8B B3  F0 00 C6 5C  90 99 F9 21  EA 45 58 FE
(V) FB F3 8C DD  70 FA FF 60  9D FE 4B AE  6B FF 98 4B
(V) FF 24 E9 69  E1 C2 DE 19  9C 8F BD 08  72 06 B0 06
(V) 02 38 17 F8  30 E4 86 53  60 4B 81 B1  F6 FF DC DD
(V) C2 F5 76 8D  99 F9 CB 87  4E 67 36 97  80 B5 17 63
(V) FF FD 15 04  3C C9 6C FB  FE FE EB AF  92 C2 5B 24
(V) 6C 0F 83 25  8E A0 11 20  C7 0B 71 EE  FF 2F 4B 46
(V) 4D 35 4B 41  49 46 41 2D  4D 45 54 45  52 0D 0A
(W) Frame checksum error
(V)   payload:
(V) 2F 4B 46 4D  35 4B 41 49  46 41 2D 4D  45 54 45 52
(V) 0D 0A 0D 0A  00 82 02 8F  30 00 00 00  00 45 1A 30
(V) A1 38 28 55  7A 57 8E 74  DA 6E 5E FF  82 D2 33 FF
(V) FE FE 1B 9D  CC E5 46 FF  67 FF 95 39  FF A6 58 FC
(V) C1 F9 0A 66  F9 FC 3F F3  17 C9 DE B2  FE 2A DE E2
(V) 4D 96 FD E9  39 FF C2 FC  5E 29 C3 98  CD 02 C1 FA
(V) 48 FE FD 8A  FE 31 02 28  1F 7C 3C E3  BF 7B FF 43
(V) FF F5 F7 EE  94 FF 23 9E  76 27 FC D7  76 E0 1C C8
(V) 79 9D 00 EB  01 3B 46 D2  FF FD 93 FF  1C C0 E3 22
(V) BB E9 42 FF  FF CD D6 FF  81 EF 9A 5F  1B FF B0 00
(V) C6 5C 90 99  53 F9 91 C5  CB FE CF 67  70 FD D8 DA
(V) B7 E0 EE FB  AE FF AE B1  FD 06 EE 80  13 80 87 A5
(V) 62 90 FC 7C  FA 5E 3F D2  C5 F8 E4 E9  61 B1 80 A8
(V) 5E 9C 8F BD  08 72 FF C0  06 B0 06 02  38 17 FF 30
(V) E4 86 D3 60  76 90 FE 99  F9 87 4E 67  FF 3C 84 FD
(V) 83 FD F5 F9  93 FF F0 36  97 80 B5 17  63 FF 28 FF
(V) 04 C9 6C DB  FC 98 FF 2F  4B 46 4D 35  4B 41 49 46
(V) 41 2D 4D 45  54 45 52 0D  0A 0D 0A 00  82 02 8F 30
(V) 00 00 00 00  45 1A 30 A1  38 28 55 7A  57 8E 74 DA
(V) 6E 5E FF 1B  9D CC FD 66  FE E3 B3 FF  94 F9 FB 77
(V) F8 CC C1 F9  0A B8 66 89  18 3F D3 F8  75 5E 16 D1
(V) A4 DE E2 FD  96 2C FC 24  2D 39 5A C4  FE 32 FF 1F
(V) FF EF CD FE  B3 FE EF FA  48 E6 FD FA  8A FE 31 25
(V) 7C FC 23 FF  C6 FB F7 43  DD FF CA 7B  12 23 76 FF
(V) E4 F7 FE 76  E0 1C C8 79  9D 00 AA E7  E8 C5 AA FF
(V) E8 A0 FF 1C  FF 38 C0 63  E2 FA BB 17  F9 CF F6 B5
(V) 81 AF 8B B3  F0 00 C6 5C  90 99 F9 21  EA 45 58 FE
(V) FB F3 8C DD  70 FA FF 60  9D FE 4B AE  6B FF 98 4B
(V) FF 24 E9 69  E1 C2 DE 19  9C 8F BD 08  72 06 B0 06
(V) 02 38 17 F8  30 E4 86 53  60 4B 81 B1  F6 FF DC DD
[... 229 lines cut ...]
(V) FA BB F7 42  C9 FF CF F6  F5 F9 21 EA  FF 58 FE CB
(V) B3 8C E7 DD  70 DA FF 60  9D FE FE EB  B1 4D E5 80
(V) 13 80 87 A5  11 F0 EC 7C  DA 5E FC D2  FF 98 24 91
(V) C2 DE FF 1E  C0 06 B0 06  02 3F 1E FC  30 E4 86 53
(V) 60 CB B1 0E  3C DC DD C2  FF 76 B7 F0  99 FE 2F 4B
(V) 46 4D 35 4B  41 49 46 41  2D 4D 45 54  45 52 0D 0A
(V) 0D 0A 00 82  02 8F 30 00  00 00 00 45  1A 30 A1 38
(V) 28 55 7A 57  8E 74 DA 6E  5E FF FA 97  FF 3E FF 1F
(V) 32 FE F9 CC  35 CE DA 9B  9F 88 FE E5  19 60 82 82
(V) D2 33 FF D2  FE 1B 9D FC  E5 FE 46 FF  67 FF 39 FB
(V) A6 58 FC C0  F9 0A B8 66  3F 28 17 C9  B2 17 DE E2
(V) 4C FC 39 C2  C4 5E FE E9  C3 98 02 B6  FF C1 FA 48
(V) FD FA 15 FE  31 02 E5 28  1F 7C 3C FF  BF 7B 43 FF
(V) 74 A0 FE AF  B1 AF F6 01  46 FF D2 FD  9E 93 FB C0
(V) 63 22 FA BB  E9 DB FF D6  B5 81 FF FA  E4 1B B0 00
(V) C6 5C 90 F9  60 E1 91 EA  FA D2 FF CF  FC FE FE F5
(V) 06 EE 80 13  80 87 A5 62  F1 90 EC 7C  5A 5E FD 24
(V) FF 2B 24 E9  FF 80 91 A8  C2 5E 19 9C  8F FD F8 72
(V) 0B C0 06 B0  06 02 38 17  FC 30 E4 86  53 60 FB 81
(V) FF F6 FD FE  FF 87 67 3C  84 E5 83 75  FF 93 DA F7
(V) F0 36 97 80  B5 63 B0 FD  28 15 04 3C  1C FB 7A EB
(V) AF 92 FC FF  FF
(E) Serial buffer overflow
(E) Serial buffer overflow
```

gskjold, 2025-09-01:
It is true, the max buffer size on 8266 have been limited due to low memory. And considering this frame reports a payload size of 0x028F (655 bytes), it is likely that the decryption actually fails because the frame does not fit in memory... But also, if you search for "02 8F" in that log, it repeats at random intervals, so I am also wondering if the correct baud/parity is used here. 

To be honest I am not completely sure what to make of it. But what I can say is that the frame starts with `2F 4B 46 4D  35 4B 41 49  46 41 2D 4D  45 54 45 52` which translates to ASCII `/KFM5KAIFA-METER`, so this is DSMR and not DLMS, at least in the start. Immediately after that header, an encryption header is present, stating that the frame size is 655 bytes. What is inside this encrypted payload is of course another question.