UtilitechAS/amsreader-firmware #1001: Decrpytion not working

Cartoon-Dude, 2025-09-09:
I finally received the key to decrypt the meter data, but it isn't doing anything.
Without putting the key: HAN: Unknown data received, check meter config
After putting the key: no message but HAN still red.
Also why is there 2 fields to put the key in? Must put 2 times the same? Let one empty? 
Thanks

Pow-P1 v2.4.7

Cartoon-Dude, 2025-09-09:
Switzerland, we already discussed there: https://github.com/UtilitechAS/amsreader-firmware/discussions/984

ArnieO, 2025-09-09:
OK, I was about to guess Austria. :-)
Try with the "Authentication key" field empty (delete the default zeros that are there as placeholder).

Use the key in the first field. 
If the provided key starts with "0x", do not include it.
After entering the key, double check it is entered correctly (all digits are there).

ArnieO, 2025-09-09:
> Also if the data is readable in the log attached before, does that mean the decryption was successful?

Sorry, I did not have time earlier to study it. 
But the answer is Yes - the decryption is successful. 
The section "DLMS frame" can be converted to somewhat readable XML format using this tool: https://www.gurux.fi/GuruxDLMSTranslator

For your first frame, this gives: 
```
<DataNotification>
  <LongInvokeIdAndPriority Value="00CB31B5" />
  <!--2025-09-09 10:38:55-->
  <DateTime Value="07E90909020A263700FF8880" />
  <NotificationBody>
    <DataValue>
      <Structure Qty="0F" >
        <UInt32 Value="01A78D92" />
        <UInt32 Value="017C742B" />
        <UInt32 Value="002B1967" />
        <UInt32 Value="010EE26D" />
        <UInt32 Value="002DDA31" />
        <UInt32 Value="00E1083C" />
        <OctetString Value="" />
        <!--2025-09-09 10:38:55-->
        <OctetString Value="07E90909020A263700FF8880" />
        <UInt32 Value="000004D3" />
        <UInt32 Value="00000000" />
        <UInt16 Value="0088" />
        <UInt16 Value="00CC" />
        <UInt16 Value="00BA" />
        <UInt16 Value="0946" />
        <UInt16 Value="0942" />
      </Structure>
    </DataValue>
  </NotificationBody>
</DataNotification>

```

This is just a listing of 0F (=15) parameters, without OBIS codes to identify them. 
In order to decode this, we need to information on the list of parameters. 

I'm not the code person here, it is possible that this list is already in our code, we have implemented quite a number of different payload formats now. ( -> Ping @gskjold !)
But the debug shows that the payload is decrypted without error messages.

So what is your situation - do you not see any data in the user interface? 
Can you provide a screenclip?

Cartoon-Dude, 2025-12-30:
Hello
Thank you very much, it seems very promising, every data seems all right.
Will observe a few days and report if any bug

Cartoon-Dude, 2025-12-31:
More exactly, the import and export power values are inverted
But still correct for the daily/hourly?

gskjold, 2026-01-01:
Please test this one:

[esp32s2.zip](https://github.com/user-attachments/files/24402605/esp32s2.zip)

closed 2026-01-08 by ArnieO