UtilitechAS/amsreader-firmware #1021: No HAN data received last 30s after upgrade to v2.4.8

XGRATION, 2025-09-30:
After upgrading to v2.4.8 the reader has started being unable to receive data from the meter - saying: "No HAN data received last 30s, single blink"

Kamstrup has reset the meter and the led is flashing blue every 6 sec., but the HAN status in dashboard is this RED.

I have tried a lot of different settings now and it has been working very well for years.. until now!

Do you have any ideas?

**Hardware information:**
 - Country: Denmark
 - Meter: Kamstrup
 - Encryption enabled: yes
 - AMS reader: Pow-K, esp32s2
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.4.8
 - MQTT: no
 - HAN baud and parity: 2400 8E1 - buffer 512
 - Temperature sensors: No
 - ENTSO-E API enabled: yes

XGRATION, 2025-09-30:
Wup wup.. now it works! Initially I used the "one-click update" clicking the cog from the upper right corner via mobile browser.
It said updating (as I remember) and after a few seconds it returned to normal screen with same version.
I tried several times with no luck. Then I uploaded it manually with the *.bin-file

But now it's working. Fantastic! Thanks.

ArnieO, 2025-09-30:
> Wup wup.. now it works! Initially I used the "one-click update" clicking the cog from the upper right corner via mobile browser. It said updating (as I remember) and after a few seconds it returned [...]

OK, I assume this was yesterday.
We had a snag on the server for a few hours Monday morning, until we discovered it.
You are the second one to have experienced this, the other person also did manual updating for the same reason.
Thank you for your help in pinpointing the issue.

ArnieO, 2025-09-30:
> Then I uploaded it manually with the *.bin-file

To be sure, as there are several ways to do an upgrade: You upload the *.bin file via the user interface?
(e.g. you did not use ESPTOOL, and uploaded over USB)?

robertsorensen, 2025-11-10:
Thanks a lot for the help and pointers.
I did reach out to my provider again this time explaining the issue described in the blog post linked above.
They did reach out to Kampstrup and within a short time (1-2 hours) my HAN interface started working again. 
So far - issue addressed but obviously it can happen again.

Appreciate the help.

ArnieO, 2025-11-10:
That is great to know. As indicated in the article this problem is very rare - I have myself only seen it once over the 6-7 years I've had reader connected to my Komatrup Omnipower.

closed 2025-11-10 by ArnieO